FreeOpcUa/python-opcua #7: Client: Problem by writing value to custom server

destogl, 2015-06-17:
Hi,

I have problem by writing value in custom server based on UA SDK. I get following error message: The server not does support writing the combination of value, status and timestamps provided.(BadWriteNotSupported). The variable is writeable for sure.

 I found a problem and it is in Timestamps, if the booth timestamp (DataValue structure) are set to 0 than works.

Should the user be enable to set custom Timestamp from client API?

oroulet, 2015-06-17:
Write a datavalue instead of a variant. Then you have full control. Isn't
it?

destogl, 2015-06-18:
Yes you are right... Would be than be better to add one more function to node.py with definition:

```
def set_value(self, datavalue)
```

or use directly set_attribute (this is a bit dirty)?

oroulet, 2015-06-18:
I did not realize that this was an issue with python-opcua!
method overloading is a but messy in python. a method definition with
my_method(*args, **kwargs) messes up documentation.
but in this particula case I think we should modify the method(and the doc
string!!) to support giving a datavalue as argument.
Do you send a patch?
just add
elig isinstance(value, DataValue):
    xxxxx

closed 2015-06-18 by destogl