FreeOpcUa/python-opcua #9: set value error

kashp, 2015-06-23:
i am trying to communicate with kepserverex and i am trying to set value of tag but getting error

code:

  client.connect()
        root = client.get_root_node()
        state = root.get_child(["0:Objects", "2:Channel1.Device1.Tag1"])
        print(state.get_value())
        state.set_value(ua.Variant([23], ua.VariantType.Int64))

error:

 File "example-client.py", line 51, in <module>
WARNING:opcua.binary_client:Received an error message type
    state.set_value(ua.Variant([23], ua.VariantType.Int64))
  File "/home/kashp/Desktop/pythonopc/python-opcua-master/opcua/node.py", line 195, in set_value
WARNING:opcua.binary_client:MessageAbort(error:StatusCode(BadInvalidArgument), reason:)
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "/home/kashp/Desktop/pythonopc/python-opcua-master/opcua/node.py", line 214, in set_attribute
    result[0].check()
  File "/home/kashp/Desktop/pythonopc/python-opcua-master/opcua/uatypes.py", line 249, in check
    raise Exception("{}({})".format(self.doc, self.name))
Exception: The server not does support writing the combination of value, status and timestamps provided.(BadWriteNotSupported)

Please help

oroulet, 2015-06-23:
yes som servers are not accepting timestamps. instead of writting a
variant, write a DataValue
dv = ua.DataValue(myvariant)
and set timestamp to 0 or None
then write it to server

kashp, 2015-06-23:
Thanks for the reply

now i modified the code as you said 

state.set_value(ua.DataValue(ua.Variant([2], ua.VariantType.UInt16)))

also tried to change the value in uatypes.py

 def __init__(self, variant=None):
        self.Encoding = 0
        if not isinstance(variant, Variant):
        variant = Variant(variant)
        self.Value = variant
        self.StatusCode = StatusCode()
        self.SourceTimestamp = 0  # DateTime()
        self.SourcePicoseconds = 0
        self.ServerTimestamp = 0  # DateTime()
        self.ServerPicoseconds = 0

but still i am getting error 
Exception: The value supplied for the attribute is not of the same type as the attribute"s value.(BadTypeMismatch)

can you please give me sample code to write the tag on the server which do not support time stamps and status
???

oroulet, 2015-06-23:
>> Exception: The value supplied for the attribute is not of the same type as the attribute"s value.(BadTypeMismatch)

This is a different error. It looks like ou are writing with the wrong variant type. Are you sure your variable is an int64?

kashp, 2015-06-24:
i am using UInt16 as a data-type,  have checked this data type from get_value() 
still facing error, please provide sample code for the servers which don’t support status and time stamp.
Thanks

oroulet, 2015-06-25:
I just commited a change to set default value of timestamps to None in DataValue. It seems some servers like keepware do not accept timstamps. This is tested with kepware server and definitely works if you give the correct VariantType.
And since you want sample with kepware:
var = objects.get_child(["2:Channel1", "2:Device1", "2:myuint"])
dv = ua.DataValue(ua.Variant(78, ua.VariantType.UInt16))
var.set_value(dv)

zerox1212, 2015-06-28:
I'm working with TOP Server (basically kepware) and I'm having a hard time reading a value using example-client.py

What are you using to get the "obects" part of:
var = objects.get_child....

I tried:
obj = root.get_child(["0:Objects", "2:Functions"])
but it always crashes and jumps to the finally code block.
Do you have a more complete example?

Thanks

oroulet, 2015-06-29:
What is the complete error?

zerox1212, 2015-06-29:
This is my first time in Python, but it looks like this is the exception upon returning from the get_child call is this:
__exception__ Class Exception The requested operation has no match to return.(BadNoMatch)

oroulet, 2015-06-29:
This means your Node does not exist... Wrong browsename or index probably
maybe path

zerox1212, 2015-06-29:
What are the rules for the get_child method? The documentation is rather vague as it only says to supply a string or list of strings. Can I just get my value by entering the entire path to my tag instead of getting the root, then getting the ojbects, then finally the tag? Sorry if these are dumb questions, I have a lot of experience with setting up OPC server software, but the inner workings are new to me. Thanks gain.

kashp, 2015-06-30:
Thanks  oroulet  
its working now 
:)

oroulet, 2015-06-30:
nice that everybody seems happy now :-)

closed 2015-06-30 by oroulet

ranjithLTTS, 2018-11-26:
Cool, saved my life

amisha218, 2019-06-04:
import time
from opcua import Client

url = "opc.tcp://131.246.76.240:4840"

client= Client(url)

client.connect()
print("Client Connected")

while True:
    Temp = client.get_node("ns=2;i=2")
    Temperature = Temp.get_value()
    print(Temperature)

    Press = client.get_node("ns=2;i=3")
    Pressure = Press.get_value()
    print(Pressure)

    TIME = client.get_node("ns=2;i=4")
    Time = TIME.get_value()
    print(Time)

    time.sleep(1)

This is my client code. I am not able to connect to my server code which is working absolutely fine. I am getting error in client.connect() line. Please do help and resolve my issue.

oroulet, 2019-06-04:
@amisha218 read a best practice about how to report a bug. Then try again

RanjithGowdaMN, 2019-06-04:
@amisha218 
i have few question: 
1. are you trying to connect to some Siemens software or any other tool
2. are you getting any error, if yes what is the error message you are getting..

you might look in to security credential once.

amisha218, 2019-06-04:
@RanjithGowdaMN 
1. No,I used UaExpert but there also I am not able able to receive my server variables inspite of that my server has been connected to it. 
Is it necessary to first connect to any Ua client application tool?

2. yeah,I am getting error in my client code which is

>>ServiceFault from server received  in response to CreateSessionRequest
Traceback (most recent call last):
  File "D:\nwc.py", line 7, in <module>
    client.connect()
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\client\client.py", line 259, in connect
    self.create_session()
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\client\client.py", line 360, in create_session
    response = self.uaclient.create_session(params)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\client\ua_client.py", line 267, in create_session
    data = self._uasocket.send_request(request)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\client\ua_client.py", line 79, in send_request
    self.check_answer(data, " in response to " + request.__class__.__name__)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\client\ua_client.py", line 88, in check_answer
    hdr.ServiceResult.check()
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python37\lib\site-packages\opcua\ua\uatypes.py", line 233, in check
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadSecurityChecksFailed: An error occurred verifying security.(BadSecurityChecksFailed)

amisha218, 2019-06-04:
@RanjithGowdaMN 
And this is my server.py code which is working fine

from opcua import Server
from random import randint
import datetime
import time
from time import sleep
server=Server()
url="opc.tcp://127.0.0.1:4840"
server.set_endpoint(url)
name="OPCUA_SIMULATION_SERVER"
addspace=server.register_namespace(name)
node=server.get_objects_node()
param=node.add_object(addspace,"Parameters")
temp=param.add_variable(addspace,"Temprature",0)
press=param.add_variable(addspace,"Pressure",0)
time=param.add_variable(addspace,"Time",0)
temp.set_writable()
press.set_writable()
time.set_writable()
print("Server started at {}".format(url))
while True:
    Temprature=randint(10,50)
    Pressure=randint(240,999)
    Time=datetime.datetime.now()
    
    print(Temprature,Pressure,Time)
    
    temp.set_value(Temprature)
    press.set_value(Pressure)
    time.set_value(Time)
    sleep(2)

amisha218, 2019-06-04:
> @amisha218 read a best practice about how to report a bug. Then try again
Please elaborate!!

amisha218, 2019-06-04:
@oroulet Please elaborate!!

keivan-mokhtarpour, 2021-09-24:
> I just commited a change to set default value of timestamps to None in DataValue. It seems some servers like keepware do not accept timstamps. This is tested with kepware server and definitely works [...]

@oroulet This one helped a lot. was looking around in the past +2 hours everywhere for a solution.