FreeOpcUa/python-opcua #11: kepserver data event

kashp, 2015-07-01:
Hi  oroulet 
I am connecting kepserver with python using python-opcua
I am polling the server for tag value change, Is there any event registration mechanism so that my python function triggered automatically as the tag value will change

Thanks

oroulet, 2015-07-01:
Look at subscription in example client. Let me know if it works with kepware

kashp, 2015-07-02:
Thanks oroulet its working thanks,
I have one more doubt, lets suppose i have registered 180 tags 
like this
sub.subscribe_data_change(myvar)

data_change(self, handle, node, val, attr) will be executed upon event receive. 
If all tags changed then this function will be called in parallel or sequential 
????

oroulet, 2015-07-02:
In current implementation there is only one thread calling callbacks so all
calls will be sequential s. One could implement a thread pool but in
python, which is anyway monoprocess, I am not sure it would help anything

oroulet, 2015-07-02:
One planned improvement is to allow subscriptions to many tags at once...
But not done yet

closed 2015-07-26 by oroulet