FreeOpcUa/python-opcua #14: subscription handler hangs when Node.get_value() called from within handler

Notre1, 2015-07-07:
I'm trying program a subscription handler to read the attribute on a separate node (separate from node the subscription is set on), but it hangs when the get_value function is called.

For example, I can successfully create a subscription handler for node (say Node A), and I can run a print command on data_change just like in the sample program.  But when I add a line to the handler to read of the attribute of a separate node (Node B), and rerun it, the program hangs.

It looks like it is hanging when BinaryClient._send_request is calling future.result because the state is "PENDING".  When I ran it through PyCharm's debugger, the last line of opcua code that runs (before it hangs) is [line 62](https://github.com/FreeOpcUa/python-opcua/blob/8fce2f7b2c29d5f73584706f7f26db93faf70d39/opcua/binary_client.py#L62) in binary_client.py.

### Example Code
If I remove the two node_b lines, it works, but with them in it just hangs:

```python
from opcua import ua, Client

class SubHandler(object):
    def data_change(self, handle, node, val, attr):
    	print("Python: New data change event", handle, node, val, attr)

    	node_b_path = ['Objects', '164 Zone 2', '164-24', 'Global', 'Part_1_Print_Label']
    	node_b_value = root.get_child(node_b_path).get_value()

client = Client("opc.tcp://internal-kepserver:49320")
client.connect()
root = client.get_root_node()

handler = SubHandler()
sub = client.create_subscription(500, handler)

node_a_path = ['Objects', '164 Zone 2', '164-24', 'Global', 'Part_1_Rack_Count_Temp']
node_a = root.get_child(node_a_path)
sub_handle = sub.subscribe_data_change(node_a)
```

jk987, 2015-10-30:
To be more precise, "hangs" means "stops doing anything and I have to kill the process", or does it mean "after one seconds it throws exception"? I suppose it is more likely the second option.

I even think that the problem is already in root.get_child() than in node.get_value().
(But it doesn't matter because both ends at BinaryClient.send_request.)
I also think that BinaryClient (or UASocketClient) doesn't parse response to this request until it is completely finished with previous response. But the previous response is not finished because the callback is waiting for get_child/get_value...

What to do with this?

My idea is that _receive() will put responses to some queue and that there will be the second thread getting items from queue and calling callbacks. This might solve the problem (and perhaps bring new problems).

oroulet, 2015-10-30:
This is behaviour by design. Added info in code and example.

closed 2015-10-30 by oroulet