FreeOpcUa/python-opcua #16: BadNoMatch only periodically

corylutton, 2015-07-21:
I have been testing against two different OPC servers, Kepware and Ignition
from Inductive Automation.  I am having issues with both.  Sometimes I get a 
BadNoMatch error when issuing a get_child command and sometimes it works just
fine.  My code is as follows for the ignition server.

    """ Testing OPCUA read/write tags """
    import opcua

    def ignition_test():
        """ Run a test against the Ignition server. """
        print("Testing Ignition OPC")
        client = opcua.Client("opc.tcp://10.9.10.209:4096")
        client.connect()
        objects = client.get_objects_node()

        myval = objects.get_child([
            "1:Devices",
            "1:[Simulator]",
            "1:_Meta:Writeable",
            "1:WriteableFloat2"])

        before = myval.get_value()
        myval.set_value(opcua.ua.Variant([1], opcua.ua.VariantType.Float))
        after = myval.get_value()

        print(before, after)

        client.disconnect()

    if __name__ == "__main__":
        ignition_test()

Running this gives the following error periodically:

    C:\Windows\system32\cmd.exe /c (python opc_test2.py)
    Testing Ignition OPC
    Traceback (most recent call last):
      File "opc_test2.py", line 28, in <module>
        ignition_test()
      File "opc_test2.py", line 16, in ignition_test
        "1:WriteableFloat2"])
      File "C:\Python34\lib\site-packages\opcua\node.py", line 285, in get_child
        result.StatusCode.check()
      File "C:\Python34\lib\site-packages\opcua\uatypes.py", line 249, in check
        raise Exception("{}({})".format(self.doc, self.name))
    Exception: The requested operation has no match to return.(BadNoMatch)

Any thoughts on what I should be looking into?  I am hoping to resolve this
and contribute back.

Thanks

oroulet, 2015-07-21:
BadNoMatch is the answer from the server. For some reasons the server does
find the requested node, I have no idea why...
do you get that error only at startup of after a random number of time? If
it is only at startup maybe we send requests too fast for these servers...
Nest step to debug this would be to use wireshark and monitor session then
look for the query called TranslateBrowsePathToNodeId that fails and the
server answer, maybe there is something to be seen there.
You can also try to use get_node instead of get_child call and get the node
directly using its NodeId. It seems many servers have issues with their
implementation of TranslateBrowsePathNodeId. Maybe this solves your issue.
Olivier

corylutton, 2015-07-24:
It seems to occur when the server has first started up, and fails with that until it just happens to successfully reads the tag the first time... then it seems to work consistently until server restart.  I'm not sure what that means but it seems I can restart and repeat the process.  UaExpert client works consistently regardless of if the server has been restarted recently.

I will do some additional tracing using wireshark as you mention and see what I can find out over the next week.

Thanks

oroulet, 2015-07-24:
Uaexpert does not do any translatenodepathtonodeid request. That is probably the difference

corylutton, 2015-07-27:
Hi Olivier,

I think I was doing it the hard way by trying to use browse path.  I can only assume it is something with the server side since it is responding and I'm not able to see anything that looks suspicious.  Switching to using get_node as you suggest it works just fine and is what I want to do anyway.

Thanks

closed 2015-07-27 by corylutton