FreeOpcUa/python-opcua #17: Timestamps

dewabe, 2015-07-26:
Hello and thank you for a create OPC UA Pythono library!
I'm using it on my thesis and I need timestamp data. I have downloaded now the latest version (just minutes ago) but still I can't find proper timestamp records.

I'm using Beckhoff TwinCAT OPC UA Server and so far there have been no issues. I can also get the timestamps as I like to (JSON format):

{
'MAIN.o_bAlarm[0]': {
    'timestamp': '2015-07-26 12:35:11.852617',
    'value': '0'
    },
and so on...
}

Timestamp is current timestamp (when the data has been checked by client), not when the value has been changed.
If I use UaExpert-tool, I can see MAIN.o_bAlarm[0].timestamp is few hours earlier as it should (as I haven't changed the value since that).

Am I doing something wrong or does the code support such a feature at all?

My application has no OPC UA connection all the time, and that's the reason I want to know how long ago these values has been changed.

Best regards,
TeemuV aka dewabe

oroulet, 2015-07-26:
Isn't simply a time zone issue? I did some simple tests and the python ua client always reports the correct timestamp. But it reports it in UTC while uaexpert seems to convert it to local time.

dewabe, 2015-07-26:
I though so first also, but unfortunately it is not. Here's two OPC UA nodes I have used for testing.
{
   "MAIN.o_bAlarm": {
      "timestamp": "2015-07-26 17:31:25.101120",
      "value": "False"
   },
   "MAIN.b": {
      "timestamp": "2015-07-26 17:31:25.112121",
      "value": "7"
   }
}

MAIN.o_bAlarm does not change, if I don't change it manually, but MAIN.b changes every second (and adds 1 to value). So on the above JSON string, there should be about 7 second difference between those two values as I just boot up my system. Of course later it will be much more..
UaExpert has "Source Timestamp" for MAIN.o_bAlarm = 20:31:25 (as my local time is +3 UTC) and for MAIN.b it is 20:31:32

I'm using following line for timestamp for "initial values" (when connection has been established, all values will be read trough OPC UA one by one and all of them will be then sent to client as JSON above):

this_node.get_data_value().SourceTimestamp)

I'm also using subscriptions but those timestamp can be handled differently anyway.

I hope this is just user error, but so far I haven't been able to found solution.

oroulet, 2015-07-26:
Can you try try to install wireshark, select the correct network interface,
set filter to opcua. Then tell us what the value of timestamp is there.

dewabe, 2015-07-26:
Wireshark gives line (for Node MAIN.o_bAlarm):
SourceTimestamp: Jul 26, 2015 22:14:28:493816900 FLE Daylight Time

Application prints value:
2015-07-26 19:14:28.493816
(same, as the program uses UTC times)

UaExpert gives:
22:11:38:748

oroulet, 2015-07-27:
By aplication you mean python-opcua?
Then it looks like we return exactly what server sends isn't it?
Do you find the timestamp uaexpert shows in packet from server in wireshark? What is the field called?

dewabe, 2015-07-27:
Looks like it is an user error. As I have played a lot with this "issue" I think there is none. The timestamp feature is little strange in OPC UA DA side. Even the server is running all the time, it doesn't store any timestamp. The timestamp is when the data has been read, not when it has been changed. Even though it would look like it in UaExpert.

The feature I am looking for is probably possible to do with historical data, or I have to find some other way to find out that info.

closed 2015-09-03 by oroulet