FreeOpcUa/python-opcua #18: How can i conect MatrikonOPC Server for Simulation

jiasondouble, 2015-07-28:
Hello. thank you for python-opcua
MatrikonOPC Server for Simulation is Run in windows2003
python-opc is client that Run in linux
the windows2003 ip is 192.168.1.110. MatrikonOPC Server for Simulation Server Name is Matrikon.OPC.Simulation.1  and the windows2003 user is OPCTest password OPC

I'm using Code in example-client.py

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)
    #logger = logging.getLogger("KeepAlive")
    #logger.setLevel(logging.DEBUG)

    client = Client("opc.tcp://OPCTest@192.168.1.110:135/Matrikon.OPC.Simulation.1")
    #client = Client("opc.tcp://localhost:4841/freeopcua/server/")
    # client = Client("opc.tcp://admin@localhost:4841/freeopcua/server/") #connect using a user

    try:
        client.connect()

but connect fail ,how can I to do

oroulet, 2015-07-28:
I have never tested pasword. but try to specify it in url. It is supposed
to be implemented...
http://www.cs.rutgers.edu/~watrous/user-pass-url.html

Does it work if logging as anonymous ( and enabling it in server) ?

jiasondouble, 2015-07-29:
thank you 
i try to use the code that  client = Client("opc.tcp://OPCTest:OPC@192.168.1.110:135/Matrikon.OPC.Simulation.1")
that also can't connect
when I run the exampl_client.py that always no reply, I don't know why.
Function “def send_hello(self, url)" in binary_client.py file 
I print the  self._lock and  future.result() .the self._lock have value is <thread.lock object at 0x1fbff030>.but future.result() is  no reply
[image]

I'm sorry my English is not good, I am from China. I hope I said above understand

jiasondouble, 2015-07-29:
another question , can  the opcua connect that MatrikonOPC ?

oroulet, 2015-07-29:
Yes I tested this server a few month ago but only using anonymous.
But I do not have access to such a server before a few weeks so I cannot
test for you.

oroulet, 2015-10-07:
duplicate of https://github.com/FreeOpcUa/python-opcua/issues/23

closed 2016-01-13 by oroulet

cscortes, 2017-08-17:
I guess I am having all kinds of badness trying to connect to Matrikon's simulation server.  I thought it would be a no brainer.   I don't think that I have it set to authentication.  How would I know?

I am using:
import opcua

c = opcua.Client("opc.tpc://localhost:135/Matrikon.OPC.Simulation.1")
c.connect()

seems simple enough, but just wont do it.  I do have Matrikons OPC explorer.  And I do see the simulation stuff fine.  I turned off the firewall just in case, I am running out of ideas.  

Ultimately I wanted to connect to Matrikon's Desktop Historian and pull items from a specific time -- is that possible with freeopcua??

zerox1212, 2017-08-18:
Can you connect to your server with UA Expert?

oroulet, 2017-08-18:
@cscortes yes this should work fine. As @zerox1212 wrote, try with uaexpert . You can also call get_endpoints() before connecting to see what endpoints are available.