FreeOpcUa/python-opcua #19: exception handling

kashp, 2015-08-04:
Hi 
How to handle OPC server Connection timeout while reading tags ore creating connections.
What are the different exceptions raised for different error.

oroulet, 2015-08-04:
No idea :-) It would be good to go though the code and check every possible
exception and raise custom exception when necessary....and document.......
patch welcome
Currently I think python-opcua returns the usual socket timeout error and
generic python exceptions if something internal happens

sscherfke, 2016-01-07:
You should *never* raise the base `Exception`.  See, e.g., https://stackoverflow.com/questions/2052390/manually-raising-throwing-an-exception-in-python for details.

oroulet, 2016-01-07:
As I wrote earlier: "patch is welcome" @sscherfke

sscherfke, 2016-01-07:
Yes, I know, but I’m still too new to opcua and this lib to provide a usefull patch. ;-)

oroulet, 2016-01-07:
I am having a short look at it, it might uncover some errors

oroulet, 2016-01-07:
https://github.com/FreeOpcUa/python-opcua/commit/e11c0588bfbedabb2adf734be71ec4b4f0d5dc79

would be better to have several exception types... but one need to think about it...

closed 2016-01-13 by oroulet