FreeOpcUa/python-opcua #21: issue

kashp, 2015-08-07:
Hi 
When Opc server restarted i got the following issue

  File "/usr/local/lib/python3.4/concurrent/futures/_base.py", line 297, in _invoke_callbacks
    callback(self)
  File "/home/kashp/PycharmProjects/gor_flip/opcua/binary_client.py", line 328, in _call_publish_callback
    response = ua.PublishResponse.from_binary(future.result())
  File "/home/kashp/PycharmProjects/gor_flip/opcua/uaprotocol_auto.py", line 10434, in from_binary
    obj.Parameters = PublishResult.from_binary(data)
  File "/home/kashp/PycharmProjects/gor_flip/opcua/uaprotocol_auto.py", line 10383, in from_binary
    obj.SubscriptionId = unpack_uatype('UInt32', data)
  File "/home/kashp/PycharmProjects/gor_flip/opcua/uatypes.py", line 135, in unpack_uatype
    return struct.unpack(fmt, data.read(size))[0]
  File "/home/kashp/PycharmProjects/gor_flip/opcua/utils.py", line 39, in read
    raise Exception("No enough data left in buffer, request for {}, we have {}".format(size, self))
Exception: No enough data left in buffer, request for 4, we have Buffer(size:0, data:b'')

oroulet, 2015-08-07:
Do you have a reproductiblr scénario or is it à one time event?

kashp, 2015-08-07:
yes when you restart OPC server 

On Client side these error started, I want to re initialize connection but due to absence of raise condition i can not handle any Broken connection exception 

I also received 

Traceback (most recent call last):
  File "/home/kashp/PycharmProjects/gor_flip/OpcuaConnection.py", line 42, in write_tag
    var = self.root.get_child(["0:Objects", read_path])
  File "/home/kashp/PycharmProjects/gor_flip/opcua/node.py", line 290, in get_child
    result = self.server.translate_browsepaths_to_nodeids([bpath])
  File "/home/kashp/PycharmProjects/gor_flip/opcua/binary_client.py", line 292, in translate_browsepaths_to_nodeids
    data = self._send_request(request)
  File "/home/kashp/PycharmProjects/gor_flip/opcua/binary_client.py", line 60, in _send_request
    self._write_socket(hdr, symhdr, seqhdr, request)
  File "/home/kashp/PycharmProjects/gor_flip/opcua/binary_client.py", line 141, in _write_socket
    self._socket.write(alle)
  File "/home/kashp/PycharmProjects/gor_flip/opcua/utils.py", line 88, in write
    self.socket.sendall(data)
BrokenPipeError: [Errno 32] Broken pipe

Traceback (most recent call last):
  File "/home/kashp/PycharmProjects/gor_flip/OpcuaConnection.py", line 43, in write_tag
    var.set_value(dv)
  File "/home/kashp/PycharmProjects/gor_flip/opcua/node.py", line 187, in set_value
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "/home/kashp/PycharmProjects/gor_flip/opcua/node.py", line 200, in set_attribute
    result[0].check()
  File "/home/kashp/PycharmProjects/gor_flip/opcua/uatypes.py", line 249, in check
    raise Exception("{}({})".format(self.doc, self.name))
Exception: The node id refers to a node that does not exist in the server address space.(BadNodeIdUnknown)

oroulet, 2015-08-07:
Obviously when I restart a server here I do not get that kind of error. Kan
you send me a reproductible scenario?

oroulet, 2015-08-07:
ok . Now I think I understand. The server break the connection and sends an
incomplete packet. I will have a look if I can add a try catch around that
or raise the exception in client thread

oroulet, 2015-08-07:
And I had one more look.....
There is no point in raising in exception there, this happens in a callback from the server. you have no method call where to raise an exception.
We could set a flag in the library but I am unsure if this is a good solution. you would nee to poll that flag....
I think if you want to detect if the connection is lost the best thing to do is to read a node value on the server at a given frequence. If the connection is lost an exception will read and you can restart client

kashp, 2015-08-08:
thanks oroultet 
I decided to restart application from supervisord on getting read or write error

But my application doesn't exit on restart the server. 

Please help. I am confused how to handle this siutuation

try:
                client = Client("opc.tcp://192.168.190.129:49320")
                client.connect()
                root = client.get_root_node()
                state = root.get_child(["0:Objects", "2:Simulation.Warehouse.Scanner.PKG_W_DATA[1,1]"])
                fast = root.get_child(["Objects", "2:Simulation.Warehouse.Scanner.PKG_C_DATA[1,1]"])
                handler = SubHandler()
                
                count=1
                while(1):
                        f.flush()
                        dv_id = ua.DataValue(ua.Variant(randint(1,10000), ua.VariantType.UInt16))
                        dv_data = ua.DataValue(ua.Variant(randint(1,10000), ua.VariantType.UInt16))
                        try:
                                state.set_value(dv_id)
                                fast.set_value(dv_data)
                                print (state.get_value(),fast.get_value())
                        except:
                                print("unable to read")
                                exit()
                        print ("count",count)
                        
                        count=count+1
                        time.sleep(4)
        except:
                print("error")
                exit()

oroulet, 2015-08-08:
I guess the keepalive thread is still running. call client.keepalive.stop()
then stop.
Does it work?

I do not see how to do this automatically in library.

Thanks for your help discovering all these things :-)

kashp, 2015-08-08:
Thanks for your help now i am able to exit the program, now i can restart the application in case of server failure.
:)

closed 2015-09-03 by oroulet