FreeOpcUa/python-opcua #25: Application hangs on Broken Pipe issue

kashp, 2015-08-26:
Hi 
Just run the below client code connect with  kepware server on the server side and restart the server again and again 
the below code will handle exceptions

but the application will stuck after some time on

Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/client.py", line 40, in run
    self.client.open_secure_channel(renew=True)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/client.py", line 145, in open_secure_channel
    result = self.bclient.open_secure_channel(params)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/binary_client.py", line 202, in open_secure_channel
    self._write_socket(hdr, asymhdr, seqhdr, request)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/binary_client.py", line 141, in _write_socket
    self._socket.write(alle)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/utils.py", line 88, in write
    self.socket.sendall(data)
BrokenPipeError: [Errno 32] Broken pipe

In utils.py at line 88 i also tried to put exception even after exception the application is hang
Please help me with this issue

def client_connect():
        try:
                global client  
                client = Client("opc.tcp://192.168.190.129:49320")
                client.connect()
                root = client.get_root_node()
                global state
                state = root.get_child(["0:Objects", "2:Simulation.Warehouse.Scanner.PKG_W_DATA[1,1]" ])
                global fast
                fast = root.get_child(["Objects", "2:Simulation.Warehouse.Scanner.PKG_C_DATA[1,1]"])
                handler = SubHandler()
                f = open('check1.txt','w')
                sub = client.create_subscription(100, handler)
                sub2 = client.create_subscription(100, handler)
                handle = sub.subscribe_data_change(state)
                handle2 = sub.subscribe_data_change(fast)

        except socket.error as e:
                if client.keepalive != None:
                        print ("stopping client")
                        client.keepalive.stop()
                        
                print ("Server Seems to be Down Retrying.......", e)
                time.sleep (5)
                main_fun()
                        
        except socket.timeout:
                print("time out error")
                main_fun()

def main_fun():
                        
       
                client_connect()
                count=1
                while(1):
                        #f.write('gin lo ji  %s \n '%count)
                        #f.flush()
                        dv_id = ua.DataValue(ua.Variant(randint(1,10000), ua.VariantType.UInt16))
                        dv_data = ua.DataValue(ua.Variant(randint(1,10000), ua.VariantType.UInt16))
                        try:
                                state.set_value(dv_id)
                                fast.set_value(dv_data)
                                print (state.get_value(),fast.get_value())
                        except Exception as e:
                                print ("teste", e)
                                if client.keepalive != None:
                                        print ("stopping client")
                                        client.keepalive.stop()
                              
                                print("unable to read")
                                time.sleep (5)
                                client_connect()
                        print ("count",count)
                        
                        count=count+1
                        time.sleep(2)
        
    
                
main_fun()

oroulet, 2015-08-26:
what is happening exactly, does the client hang? or do you get a broken
pipe in loop?

btw your loop looks a bit over complex.
Wy not:

while True:
    try:
        client_connect()
        xxxx
        xxx

 except Exception as e:
    if client.keepalive != None:
        client.keepalive.stop()
    time.sleep(5)

    #no need to reconnect in exception loop

an your use of global might lead to issues. better pass client as
argument to client_connect()

or use a class

kashp, 2015-08-27:
Hi oroulet

The above mentioned code works fine if you ll restart the server one or two time, even when you will shut-down the OPC server but when you frequently re initialize the code again and again then i am getting this Broken pipe error and after this error nothing happens even application dont crash it get stuck. 
From the traceback i can see some thread has got an Exception in your code. 
Please crash the application or close the whole application on this Broken Pipe error, I am uable to do so.

I ll get Broken Pipe error on one time and after that nothing will work

Exception in thread Thread-12:
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/client.py", line 40, in run
    self.client.open_secure_channel(renew=True)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/client.py", line 145, in open_secure_channel
    result = self.bclient.open_secure_channel(params)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/binary_client.py", line 202, in open_secure_channel
    self._write_socket(hdr, asymhdr, seqhdr, request)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/binary_client.py", line 141, in _write_socket
    self._socket.write(alle)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/utils.py", line 88, in write
    self.socket.sendall(data)
BrokenPipeError: [Errno 32] Broken pipe

Thanks 
Kashp

kashp, 2015-08-27:
hi oroulet
When I pressed ctrl -c then i get traceback as
Traceback (most recent call last):
  File "example-client.py", line 156, in <module>
    main_fun()      
  File "example-client.py", line 134, in main_fun
    state.set_value(dv_id)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/node.py", line 187, in set_value
    self.set_attribute(ua.AttributeIds.Value, datavalue)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/node.py", line 199, in set_attribute
    result = self.server.write(params)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/binary_client.py", line 258, in write
    data = self._send_request(request)
  File "/home/kashp/Desktop/pythonopc/pythonopcua2/python-opcua-master/opcua/binary_client.py", line 62, in _send_request
    data = future.result()
  File "/usr/local/lib/python3.4/concurrent/futures/_base.py", line 397, in result
    self._condition.wait(timeout)
  File "/usr/local/lib/python3.4/threading.py", line 290, in wait
    waiter.acquire()

can you please check is this the reason to stuck the application

kashp, 2015-08-27:
hi oroulet  
Error is getting fixed by changing 

binary_client.py
<             data = future.result()
---
>             data = future.result(timeout=1)

is there any side effect of this change
or what alternate can be used

oroulet, 2015-08-27:
Hi,
I had a look and this looks like a good solution!
Can you add a member timeout to BinaryClient class. set it to 1 as default and put it as parameter in future.result()
Then send a pull request :-)
I can also do it if you do not want your name in commit :-)

kashp, 2015-08-28:
hi oroulet 
i tried to push my commit to ur repo 
but it gives error

remote: Permission to FreeOpcUa/python-opcua.git denied to kashp.
fatal: unable to access 'https://github.com/FreeOpcUa/python-opcua/': The requested URL returned error: 403

i created new branch and pushing that one for the pull request

oroulet, 2015-08-28:
YEs you need to create pull request from your own branch

closed 2015-10-30 by oroulet