FreeOpcUa/python-opcua #26: GPL vs LGPL

abrown43, 2015-08-26:
The website http://freeopcua.github.io/ states LGPL http://freeopcua.github.io/.  The python setup.py states GPLv3 https://github.com/FreeOpcUa/python-opcua/search?utf8=%E2%9C%93&q=license.  

I would like to use it and happy to contribute back any fixes we find, however, I can't use GPL but can use LGPL.

Are you willing to clarify this?

Best

oroulet, 2015-08-26:
>> The website http://freeopcua.github.io/ states LGPL
http://freeopcua.github.io/. The python setup.py states GPLv3

This is correct the C++ code is LGPL and the python code is GPL. I thought
that LGPL did not make sense with python but googling around now I see that
their might be issues.
Is someone who has contributed to python-opcua agains relicencing from GPL
to LGPL?

oroulet, 2015-09-03:
Now is the code lgpl. I am now waiting for contribution :-)

closed 2015-09-03 by oroulet

abrown43, 2015-09-03:
Thank you!  We will start using it and contributing back.  Best, Al

Al Brown
http://www.albrown.com/ | al@albrown.com | 714 251 6441