FreeOpcUa/python-opcua #30: Disconnection problem : server with python-opcua, client with C# Unified Automation SDK

AKEOPLUS-boris-bocquet, 2015-09-21:
Hello everyone,

I am currently experiencing weird behaviour concerning the disconnection of a client from a server.

Here is a short description : 

OPC-UA server
Running on Ubuntu 14.04 (LTS) (unity DE), 64 bits
Implemented in python using pythin-opcua (https://github.com/FreeOpcUa/python-opcua)
IP address : 144.3.82.70:4841
Server URL : opc.tcp://141.3.82.70:4841/freeopcua/server/

OPC-UA client
Running on Microsoft windows 7 64bits
Implemented in C# using the Unified Automation SDK .net (https://www.unified-automation.com/downloads/opc-ua-development.html)
Ip address : 144.3.82.76

I am unable to disconnect. BUT with Unified Automation's UAexpert, I can connect and disconnect. So I spied with wireshark, and there is actually a difference.

Can you help me? 

More data (including the wireshark capture files) can be downloaded here : 
https://www.dropbox.com/s/abw8wi7aq8rvz2h/TroublesDisconnectionOpcUa.zip?dl=0

Thanks in advance

Cheers

Boris Bocquet
Application Engineer –Vision & Robotics
AKEO PLUS SAS

AKEOPLUS-boris-bocquet, 2015-09-21:
New upload from wireshark. Also containing the connection / disconnection
https://www.dropbox.com/s/n6yfhz8d9lyy9tz/EnvironmentServer.zip?dl=0

oroulet, 2015-09-21:
Hi,

thanks for the report.

in your wireshark log  right click a line and select decode as ... opcua
then you will be able to see more easily what is send or not.

what does the server say when the client tries to disconnect?

Can you send a simple compile program I can run on windows to connect to a
server?

AKEOPLUS-boris-bocquet, 2015-09-21:
Hi.

I checked with « OPCUA » formatting in wireshark.

Here are the main differences :

With UA expert :
There are a couple of requests more
And the ReturnDiagnostics of CloseSessionRequest is == 0
[cid:image001.png@01D0F48E.D45AD3C0]

Whereas on the C# client, there are less request, and the ReturnDiagnostics of CloseSessionRequest is == 99

[cid:image002.png@01D0F48E.D45AD3C0]

Does it help you?

My may have missed a couple of details… But I did spend some time on cross checking

I will send you a compiled version of my client (within 5 minutes)

Are you running on a 64 bits windows?

Many thanks for your support

BR

Boris

AKEOPLUS-boris-bocquet, 2015-09-21:
Please find attached 2 builds of executable OpcUa client

A debug and a release build

Please note that this code is not yet officially open source (soon I hope), so please do not diffuse it.

BR

Boris

oroulet, 2015-09-21:
Yes I have 64 bit. Send the exe with drop box otherwise will end up in an
anti virus filter

AKEOPLUS-boris-bocquet, 2015-09-21:
Yes please : 
https://www.dropbox.com/s/anlkas3wvs682vk/SkillPro_OpcUaClientCsharp.zip?dl=0

oroulet, 2015-09-21:
Had a short look. Looks like your client is not closing socket. I am not
sure what is the correct behavior here. Server could close socket but some
clients expect to be able to send a close_secure_channel after
close_session. While your client obviously expect the server to close
socket right after close_session... Or maybe you should close socket
yourself in c#?

AKEOPLUS-boris-bocquet, 2015-09-21:
Yes I had this idea too.
But no socket closing method is available
So I tried to close socket myself by completely disposing the objects, and calling the garbage collector. Even with that trial I was still connected to the server (at least the server did not noticed my disconnection).

Only when I shutdown (kill) the application, the server notices that the client “died”.

Any other proposal?

BR

Boris

oroulet, 2015-09-22:
Strange, the SDK api should let you close socket. Anyway I looked at the
spec and server should close socket after close channel request. So I added
this and now your example works fine. Never saw this bug since all clients
seem to close socket themselves. Committing tonight

But you had another issue with connecting. Could you send me either
wireshark log or better exe to test this?

AKEOPLUS-boris-bocquet, 2015-09-22:
Thank you for your help.

Concerning the connection, I added new executables in the dropbox : SkillPro_OpcUaClientCsharp_connectionNotWorking_securityNone.zip

Also there is the UA SDK in the drop pox, if you want to play with.

oroulet, 2015-09-22:
fix commited for socket. can you send  a link to new zip file? I am not a
dropbox expert

AKEOPLUS-boris-bocquet, 2015-09-22:
Oups sorry.

Here the link of the entire folder

https://www.dropbox.com/sh/ita3wpuc6mdhywn/AAC0kSlhvPPRkn135lxNMfKBa?dl=0

oroulet, 2015-09-22:
Hi again, I tested last exe and I easily connect to python server. Does it
fail at your site?

AKEOPLUS-boris-bocquet, 2015-09-22:
Yes it does. I just rechecked again (by downloading what is on the dropbox) and both debug and release builds are unable to properly connect to the OPC-UA server.

Are your sure you are using the executable in "SkillPro_OpcUaClientCsharp_connectionNotWorking_securityNone.zip" (sorry for the stupid question).

If yes, then this probably mean that my colleague has not set some parameters properly on the server.

oroulet, 2015-09-23:
I tried again with both samples (debug and release) and both connect. Can you make a wireshark log and spot the difference in the create_secure channel or create_session packet?. I also looked at the code and I could not see any filter in open_secure_channel, the server should accept whatever shit client send,

AKEOPLUS-boris-bocquet, 2015-09-23:
So I did again the experience with both working and not working versions. The wireshark capture files are in the dropbox

https://www.dropbox.com/sh/ita3wpuc6mdhywn/AAC0kSlhvPPRkn135lxNMfKBa?dl=0

In the NOT working scenario, we can see that the server closes the secure channel after the GetEndPointsRequest/Response

AKEOPLUS-boris-bocquet, 2015-10-01:
Dear Oroulet, did you find the reason of the connection issue (with security policy none)? 

As a reminder, I did find a solution by using the string "http://opcfoundation.org/UA/SecurityPolicy#None"

just as following : 

```C#
public const string SECURITY_POLICY_NONE_URI = "http://opcfoundation.org/UA/SecurityPolicy#None";
EndpointDescription DescriptionOfEndpoint = new EndpointDescription(server_URL);
DescriptionOfEndpoint.SecurityPolicyUri = SecurityPolicyUri;
m_session.Connect(DescriptionOfEndpoint, _SettingsRequest);
```

oroulet, 2015-10-14:
I looked at your records but in the xx_working record there is no GetEndpointRequest call from you client. Is it always the case?

AKEOPLUS-boris-bocquet, 2015-10-19:
I usually don't call this method because I know the endpoint beforeend. So you should not look into this direction : the error is probably somewhere else.

Cheers

oroulet, 2015-10-19:
ok. But to understand what is happening we need a  log for a client trying to connect in an identical manner. In the 2 logs, the calls are not the same. one calls GetEndpointRequest while the other does not

AKEOPLUS-boris-bocquet, 2015-10-19:
Well, I am sorry but I don't currently have the OPC-UA server on my laptop. I will ask some colleagues. If you don't have any answers within one week, just consider the issue as closed.

Cheers

closed 2016-01-13 by oroulet