FreeOpcUa/python-opcua #34: Calling Methods from UaExpert

destogl, 2015-09-30:
To be able to debug/test my application base on this stack I would like to call Methods from UaExpert. Usually, UaExpert generates from based on input and output parameters where the values can be entered. With the python-opcua stack this is not the case. In the figure below you can see on the left side multiply method from example server and on the right call of (completely different) method on UA sdk server.
 
Last two days I am search for solution, I tried add names and descriptions (also play with references) but nothing helped. Somebody has idea?

[image]

oroulet, 2015-09-30:
interresting.... I thought it worked...I see the method GetMonitoredItems
in ObjectTypes/BaseObjectTypes/ServerType also has the same issue. I think
there is not other way than starting wireshark and comparing the server
answer for a browse request on the method and ta read of all attributes on
InputArguments and the method. Must be a difference somewhere......I
thought maybe because Name value of arg was missing. But it is not...

destogl, 2015-09-30:
So I found the problem (hopefully the right one :) )

It happens by searching element by node ([_find_element_in_node function in address_space.py file](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/address_space.py#L158)). The reason is that in *QualifiedName*  sent from UaExpert *Namespace Index==0* (TargetName in data structure see figure) and the [checking](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/address_space.py#L158) to be 2.

[image]

For now I will just check the BrowseName. I have also following question: Should the TargetName inherit the NamespaceIndex?

destogl, 2015-09-30:
Solved in #35

oroulet, 2015-09-30:
I do not understand, I do not see any translateBrowsePathToNodeId request in wireshark before calling a method from uaexpert

oroulet, 2015-09-30:
I think I found a much more probable reason for the error.
ArrayDimension is 0 in freeopcua server for method while ArrayDimentsino seems to be the number of arguments in prosys server. Looks like ArrayDimensino must be set up when Value is an extensionObject of a given size. Maybe it should also be set for any array... something to look at

destogl, 2015-10-01:
[Here](https://iirobcloud.ira.uka.de/index.php/s/vlNU4HuKvZaLYKw) you can find video showing the call of TranslateBrowsePathToNodeId when the MethodCall is made from UaExpert. Also there are logs from wireshark. In the logs you can see that answer ist still not exactly correct, since has too long array inside (test with other server returned array with 2 elements) - this explains your second comment probably. Calling of TranslateBrowsePathToNodeId makes since if you check description of Method NodeClass (Part 3, Page 30, Table 10) since *InputArgument* and *OutputArgument* Nodes has to be found.

Also you can see in request that NamespaceIndex of TargetName is 0, which makes kind a sense if you check description in Part4, Page 149, Table 163:
```
A sequence of References and BrowseNames to follow. This structure is defined
in-line with the following indented items.
Each element in the sequence is processed by finding the targets and then using
those targets as the starting nodes for the next element. The targets of the final
element are the target of the RelativePath.
```

destogl, 2015-10-02:
Closing, since there is discussion in #35

closed 2015-10-02 by destogl