FreeOpcUa/python-opcua #40: Strong Typed Variables Don't Pass Validation

zerox1212, 2015-10-04:
When another OPC UA server is the client to python opcua "Strong Typed" variables don't pass validation.

I am using TOP Server OPC UA client to my python OPC UA server and it only adds variables with no ua type.

From the example-server, this will be thrown by TOP Server:
```Date        Time         Level        Source                               Event
10/4/2015    5:09:47 PM  Warning      OPC UA Client                        Tag 'myStronglytTypedVariable' on device 'Channel1.Device1' was not imported. The item failed validation.
```

I don't know what validation is taking place? Is there any way to troubleshoot this?

oroulet, 2015-10-05:
as long as this product does not report a better error, it is hard to find
out what is happening..

oroulet, 2015-10-30:
I installed kepware and got something similar, not all variables are accepted by kepware client....

kevincolyar, 2015-12-28:
@zerox1212 Having the same issue.  If I set the type, float for example:

```python
return ua.Variant(float(value), ua.VariantType.Float)
```
 my TOP server sets it as a Float Array.

Have you found a good workaround without having to manually set the datatypes in TOP Server?

oroulet, 2015-12-28:
can it be an issue with ValueRank attribute? try set it to -2, -1, 0, 1, etc

kevincolyar, 2015-12-28:
Where would I set the ValueRank?  Here's what I'm doing now:

```python
folder.add_variable(self.namespace, name, ua.Variant(float(value), ua.VariantType.Float)
```

oroulet, 2015-12-28:
Your call return a node object. Then call node. Set_attribute
(ua.attributeids.valuerank, data value(-1, ua.varianttype.XXX))

kevincolyar, 2015-12-28:
Thanks, that was it:

```python
node.set_attribute(ua.AttributeIds.ValueRank, ua.DataValue(-1))
```

oroulet, 2015-12-29:
Does it also work with -2?
-1 means scalar while -2 can be anything

kevincolyar, 2015-12-29:
-1 : Correctly results in Float (Timestamp hour is offset)
-2 : Incorrectly results in Float Array (Timestamp is correct)

oroulet, 2015-12-29:
ok then we know it. I pushed a change to master
https://github.com/FreeOpcUa/python-opcua/commit/64e049595105c46600316e6ff2e127a329c6a219
. Now we set better default value for valuerank automtically at at variable
creation

kevincolyar, 2015-12-29:
Thanks so much.  Works perfectly and the timestamps are correct now.

closed 2016-01-02 by oroulet