FreeOpcUa/python-opcua #41: Standard_Address_Space_Part5

zerox1212, 2015-10-05:
I'm trying to install python opcua on an Arduino Yun (python 2.7).

When I do this:
```pip install -v freeopcua```
All dependencies install ok, but when the installation reaches this point:
```byte-compiling /usr/lib/python2.7/site-packages/opcua/standard_address_space_part5.py to standard_address_space_part5.pyc```
It locks up. When I look at all the other "Standard_Address_Space" files they are very small, but part 5 is large at 1.3mb.

Is part 5 required for the OPCUA server to run? I really want to figure out how to get this installed on my Yun.

oroulet, 2015-10-05:
spec says it should be there.
look at internal_server.py around line 49. default is to load address space
but you can comment it out or load it directly from xml (xml is in source
under schema directory) or even load a python dump file (generated by
runnning generate_address_space.py in schema dir)
let me know how it goes and how long each method take on a raspberry pi

zerox1212, 2015-10-05:
Do you know if there is a way to stop setup.py (or is it pip?) from compiling the address space .py files to .pyc? I don't know how to manually install python opcua so that I can try what you mention above.

oroulet, 2015-10-06:
>> p setup.py (or is it pip?) from compiling
might be tricky

but

git clone xxxxxxx/python-opcua
cd python.opcua
vi opcua/internal-server.py # to edit whatever you want
python example-server.py

and when you are happy
python setup.py install

hard to make it easier...

zerox1212, 2015-10-10:
So I found out what was happening.
1. pip can't install freeopcua because when byte compiling the file I mentioned the OS runs out of memory (Yun only has 64mb).
2. I tried the modifications you mentioned, and the server tries to start, but again the OS runs out of memory.

I'll try my project on a Raspberry Pi or Beaglebone Black next and let you know. Thanks.

oroulet, 2015-10-11:
I just tested with raspberry pi version B and the server is slow to start but run fine without any changes. using dump file should make startup much quicker

zerox1212, 2015-10-12:
I just loaded up the example OPC UA server on Beaglebone black (Python 2.7). It takes about 10 or 15 seconds to start the server, but after that it looks OK. I was able to connect to it with an OPC UA test client no problem. This could be a great option for very simple data connection between smart devices!

closed 2015-10-30 by oroulet