FreeOpcUa/python-opcua #42: client.connect() activate.session()

mertz3hack, 2015-10-06:
Looking for help understanding a connection issue.
I am trying to connect to a machine hosted opc ua server using
    <code>client = Client("opc.tcp://user:password@192.168.15.30:4880/Arburg")</code>

I am getting the following error and am not sure if this is something I am doing or if this is not supported yet.

<code>Traceback (most recent call last):
  File "example-client.py", line 41, in <module>
    client.connect()
  File "C:\opcua\opcua\client.py", line 111, in connect
    self.activate_session()
  File "C:\opcua\opcua\client.py", line 199, in activate_session
    params.UserIdentityToken.Password = bytes(self.server_url.password)
TypeError: string argument without an encoding
WARNING:opcua.binary_client:Received an error message type
WARNING:opcua.binary_client:MessageAbort(error:StatusCode(BadInvalidArgument), reason:)
</code>

oroulet, 2015-10-06:
Yes, there is an issue with the password implementation. I do not know how to encode it. it is probably just a matter of reading the doc , though.... so help welcome here

mertz3hack, 2015-10-07:
Should this issue be combined into the other issue?

oroulet, 2015-10-07:
yes there is already an issue on this https://github.com/FreeOpcUa/python-opcua/issues/23 . you are welcome to try to solve this :-)
look at line 199 in client.py. We need to encrypt password and set correct encryption in line under.

To find out what should be there you can read the opc-ua specification (But I could not find anything) and/or start wireshark, set filter type to opcua,  and connect with a client supporting encryption (uaexpert for example). then do the same with python-opcua and look at the difference in the ActivateSession message

oroulet, 2015-10-07:
btw what server are you using?

mertz3hack, 2015-10-09:
I am using an Arburg Injection Molding Machine that hosts its own opc-ua server

oroulet, 2015-10-09:
That's interesting!. Do you manage to connect if you disable password?

oroulet, 2015-10-10:
yesterday I registered a session between a client and a prosys server:
server return 2 supported PolicyId: username_basic256 and username_basic128
and security policy URI is set to http://opcfoundation.org/UA/SecurityPolicy#Basic256
when sending password in activateSession it refers to following EncryptionAlgorythm
http://www.w3.org/2001/04/xmlenc#rsa-oaep
I have no idea where the key for the encryption is taken from ...

mertz3hack, 2015-10-12:
I am not able to disable the password
I will try and watch a session to see how it compares

oroulet, 2015-12-28:
password and encryption is supposed to be implemented for client now. Can someone test and resport?

mertz3hack, 2015-12-28:
I am having the worst time trying to get python with pycrypto setup on my new computer.  I will be able to test this out on Wednesday assuming I get pycrypto working.

oroulet, 2015-12-29:
Master should not use pycrypto but cryptography module... But I am not sure
all patch have been merged

oroulet, 2015-12-29:
I merged the last patch removing pycrypto ( I hope I did not break to much things for you @alkor !!!). Everything is in place for password encryption, but it might no be implemented yet. I am waiting for @alkor to merge the rest of his crypto stuff

mertz3hack, 2015-12-30:
No errors about pycrypto anymore.

[image]
[image]

oroulet, 2016-01-02:
Now I tested password encryption and it works against prosys server. Thanks @alkor

alkor, 2016-01-02:
That's great!

closed 2016-01-13 by oroulet

joaopmrod, 2016-02-18:
I'm trying to connect using this: opc.tcp://admin:password@localhost:4096 and  I get this error:

showing error:  name 'uri' is not defined 1.

If I enable anonymous access I can do it.

mertz3hack, 2016-02-18:
@joaopmrod is the error you are getting similar to the one in this issue #135 
unfortunately in that specific case I am not able to enable anonymous access