FreeOpcUa/python-opcua #47: Wrong SecurityTokenId when closing SecureChannell?

jk987, 2015-10-25:
I noticed in log from Kepware OPC Server that when my client app disconnects then session is deleted but SecureChannel isn't closed. I compared it with UaExpert which disconnects from Kepware correctly. I watched into Wireshark that my app is sending CloseSecureChannelRequest with SecurityTokenId = 0 (while eg. in CloseSessionRequest there is SecurityTokenId = 1) and Kepware returns ErrorMessage with BadInvalidArgument.

The following modification of `BinaryClient.close_secure_channel()` fixes the problem (inspired from `_send_request()`):
```python
        #symhdr = ua.SymmetricAlgorithmHeader()
        symhdr = self._create_sym_algo_header()
```

jk987, 2015-10-25:
Only small notice: During today tests I'm also getting feeling that if I try to read data at the same time when SecureChannel is being automacially renewed then something wrong with the channel happens (like if Kepware OPC server kills the socket from some reason?). But I have no luck to reproduce it and catch it by Wireshark.

oroulet, 2015-10-26:
This sound obvious so I committed it in dev branch

closed 2015-10-26 by oroulet