FreeOpcUa/python-opcua #48: ReadRequest and renew SecureChannel collision...

jk987, 2015-10-26:
I finally caught it (I noticed it in issue #47 yesterday):

http://ulozto.net/xWGF411f/test-ua2-zip
password: ua1234

There is a log and pcapng. In log there is interesting time 2015-10-26 08:57:19.983. In pcapng there is interesting time 2063 with filter `tcp.stream eq 2`.

I have another things to do today but it seems to me that RequestResponse and OpenSecureChannel have switched SequenceNumber. And Kepware OPC server seems not to like it. I haven't investigate it much yet, but lock is bit different in `BinaryClient.open_secure_channel()` than in `BinaryClient._send_request()`.

oroulet, 2015-10-26:
thanks for all the bug reports. I will have a look a soon as I get 2 minutes

oroulet, 2015-10-26:
I tried to have a look at it but it looks there are several channels opened at the same time so it is hard to understand what is happening. Do you start several client at the same time?

jk987, 2015-10-26:
Yes, usually in my app there are 4 clients at the same time. (Two devices "Channel1mc.EVA" and "Channel2mc.HC" and for each device two halves "WP1" and "WP2". In my log I often call it EVA1, EVA2, HC1, HC2.) Like I wrote, in Wireshark you should filter `tcp.stream eq 2` (filter on toolbar or Menu ~ Analyze ~ Display Filters). It is the client reading tags with the name starting on "Channel2mc.HC.WP2..." (so "HC2").
It really wasn't easy to find. :)

jk987, 2015-10-29:
Already 3 days I'm running my app with this modification:
```python
    def open_secure_channel(self, params):
        self.logger.info("open_secure_channel")
        with self._lock:
            request = ua.OpenSecureChannelRequest()
            request.Parameters = params
            request.RequestHeader = self._create_request_header()

            hdr = ua.Header(ua.MessageType.SecureOpen, ua.ChunkType.Single, self._security_token.ChannelId)
            asymhdr = ua.AsymmetricAlgorithmHeader()
            seqhdr = self._create_sequence_header()

            future = Future()
            self._callbackmap[seqhdr.RequestId] = future
            self._write_socket(hdr, asymhdr, seqhdr, request)

        response = ua.OpenSecureChannelResponse.from_binary(future.result(self.timeout))
        response.ResponseHeader.ServiceResult.check()
        self._security_token = response.Parameters.SecurityToken
        return response.Parameters
```
and it seems to run OK for me.

But I did this modification intiutively accroding to BinaryClient._send_request() without any deeper understading the headers logic.

oroulet, 2015-10-29:
you are right. a lock is missing here and in close_secure_channel too. Can
you create a pull request? So you will get the immense privilege of having
your name as author :-)

jk987, 2015-10-29:
Sorry, but I can't. I'm not very familiar with Git.
(I think that my little modifications don't entitle me to be named somewhere. You are doing the big job, not me. For me it is the most important that I can tell my customer that my app finally works.)

oroulet, 2015-10-29:
As you want but you should really learn git. It makes programming much
easier

oroulet, 2015-10-30:
fixed in https://github.com/FreeOpcUa/python-opcua/commit/c4425c23d5d8e8d843c3b0e7db68db7283473718

closed 2015-10-30 by oroulet