FreeOpcUa/python-opcua #51: Malformed OpenSecureChannelResponse

Stasik0, 2015-10-29:
Hey guys,

we got a user who tested our open62541 client with pyhton server (and it did not work).

As far as I see in wireshark, the OpenSecureChannel response is malformed (and our client returns a decode error). However UaExpert seems to ignore this problem and just carries on, however a malformed packet is there.

The wireshark capture can be found here: http://s000.tinyupload.com/index.php?file_id=50188691273227621360

A screenshot is shown below:

[image]

Any idea what's the reason?

oroulet, 2015-10-29:
I downloaded the file and I cannot see any malformed packet. What version of python-opcua do you use? (But nobody has touched parsing files for  several months so it should not change much),  and what version of wireshark? I have wireshark 1.12.7
it looks like the security token is wrong on your screenshot. Wireshark says channelId 16 and tokenId 15 in the log I downloaded from you in second response

Stasik0, 2015-10-29:
just cloned master
From what I see the channel and token id have a different byte-order so 268435456 is actually a flipped 16, i got wireshark 1.10.6, so i will try to update wireshark, standby

oroulet, 2015-10-29:
Are you on ARM? I only tested on an intel prosessor. I tested a few months ago on a raspberry and everything seemed to work. But I may have missed somehting...

Stasik0, 2015-10-29:
nope, intel i5 here.... new wireshark shows no errors (this is spooky)... I will probably have to dig into our code :(

oroulet, 2015-10-29:
>> I will probably have to dig into our code :(
Exactly opposite feeling here :-)

Stasik0, 2015-10-29:
okay, it was our problem... due to a bug: https://github.com/acplt/open62541/commit/bed4376400e14601195a2c474bcaf9238d0f3759, we did not decode the inner status code, and introduced a shift into the decoding which resulted in martian string lengths. thank you for help

closed 2015-10-29 by Stasik0