FreeOpcUa/python-opcua #53: XML Importer Not Setting "Value" with Float Datatype

zerox1212, 2015-10-29:
I have been trying to trouble shoot this for the last hour or so but I can't figure it out.

If I try to use this XML code with the server import funciton:
```xml
  <UAVariable NodeId="i=30004" BrowseName="MyXMLVariable" DataType="Float">
    <References>
      <Reference ReferenceType="HasTypeDefinition">i=69</Reference>
      <Reference ReferenceType="Organizes" IsForward="false">i=30001</Reference>
    </References>
    <Value>
      <Float>1.23</Float>
    </Value>
  </UAVariable>
```
The node is created, but the value is Null, even though a value is specified. When I step through the code the xmlimporter.obj.value is never assigned. If the type is UInt32 it works correctly.

zerox1212, 2015-10-29:
Update:
I found the problem.
```python
    def _parse_value(self, el, obj):
        for val in el:
            ntag = self._retag.match(val.tag).groups()[1]
            obj.valuetype = ntag
            if ntag in ("Int32", "UInt32"):
                obj.value = int(val.text)
            elif ntag in ('ByteString', 'String'):
                mytext = val.text.replace('\n', '').replace('\r', '')
                # obj.value.append('b"{}"'.format(mytext))
                obj.value = mytext
            elif ntag == "ListOfExtensionObject":
                self.logger.info("Value type not implemented: %s", ntag)
            elif ntag == "ListOfLocalizedText":
                self.logger.info("Value type not implemented: %s", ntag)
            else:
                self.logger.info("Value type not implemented: %s", ntag)
```
Should probably throw a TODO in that code since the XML importer doesn't even support the OPC UA base variants.

zerox1212, 2015-10-29:
So, another part of this issue with the importer.

The code that checks for a value won't write a "0.0" to the variable node.
```python
    def add_variable(self, obj):
        ...
        if obj.value:
            attrs.Value = ua.Variant(obj.value, getattr(ua.VariantType, obj.valuetype))
```

This if statement should probably check for null's instead.

oroulet, 2015-10-29:
Yes parsing of values is half implemented. There is a bug for it. Would be
nice if you could fix it.

zerox1212, 2015-10-30:
After using the XML to import nodes, how can I access the nodes created by the server? For example if I want to subscribe to one.
server.get_node returns a node object even if the node doesn't exist.
Is there a way to get all the nodes that exist on the server?

oroulet, 2015-10-30:
get_node just constucts an Node object. If you want if the node is valid call a method from it.
it looks like you create a node which is child of node i=30001. so get that node and call get_children og get_child 
> s there a way to get all the nodes that exist on the server?
A server has so many nodes, you do not want to do that....

zerox1212, 2015-12-09:
I got some time to add a few fixes to the xml importer. I submitted a pull request. I hope I did it correct, it's my first pull request ever.

zerox1212, 2015-12-09:
Quick question:
How can I import the NodeID as String instead of Numeric? NodeID 30,000 isn't very nice.

oroulet, 2015-12-10:
<UAObject NodeId="s=mystringid" BrowseName="MyXMLFolder"  >

closed 2015-12-10 by oroulet