FreeOpcUa/python-opcua #55: BrokenPipeError

kashp, 2015-11-17:
Hi 

I registered almost 300 tags with Kepserver and receiving the continuous value and making separate thread for each received value.
But my system crashed due to below error:

Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner
    self.run()
  File "./opcua/client.py", line 40, in run
    self.client.open_secure_channel(renew=True)
  File "./opcua/client.py", line 145, in open_secure_channel
    result = self.bclient.open_secure_channel(params)
  File "./opcua/binary_client.py", line 203, in open_secure_channel
    self._write_socket(hdr, asymhdr, seqhdr, request)
  File "./opcua/binary_client.py", line 142, in _write_socket
    self._socket.write(alle)
  File "./opcua/utils.py", line 88, in write
    self.socket.sendall(data)
BrokenPipeError: [Errno 32] Broken pipe

Please help me to handle such cases.

oroulet, 2015-11-17:
The server has closed the connection. But why it has done it is hard to find out. How many client objects to use create? and how do yo subscribe to these nodes? with one call or 300?
You probably need to fire up wireshark and monitor your session until crash.

Btw: Do you really create a new thread for every node? That sounds crazy

kashp, 2015-11-17:
I have subscribed one function with all 300 tags with single client object, upon receiving tag value i create the thread to make it asynchronous.
I just want any exception so that i can restart the connection in such scenario.

oroulet, 2015-11-17:
if you subscribe to 300 values then you should not create a thread for each of them. Create one thread for all nodes.
Then if you want to make sure that the client is connected you should have one more thread that read a node and reconnects if reading fails. this is so common I should write an example... one day...

kashp, 2015-11-22:
Hi Oroulet 
I started one more thread to reconnect if reading failed. But now i got following error:

File "/usr/lib/python3.4/threading.py", line 920, in _bootstrap_inner                                                                [25/1972]
    self.run()
  File "./opcua/client.py", line 40, in run
    self.client.open_secure_channel(renew=True)
  File "./opcua/client.py", line 145, in open_secure_channel
    result = self.bclient.open_secure_channel(params)
  File "./opcua/binary_client.py", line 205, in open_secure_channel
    response = ua.OpenSecureChannelResponse.from_binary(future.result(self.timeout))
  File "/usr/lib/python3.4/concurrent/futures/_base.py", line 404, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError

Please help me to resolve this error