FreeOpcUa/python-opcua #57: event better command line tools

oroulet, 2015-11-17:
This bug report is for brainstorming and ideas for better command line after initial work from @alkor 

discover.py
* maybe we should print which server we are connected to in output? (ls commands do not do that!)
* In addition the default discovery port is 4840 (for binary), maybe we should have a default port and even a default url using localhost?
* maybe accept only hostname: "discoverua myhost" using default port

ls.py:
* should add support for browsename and browsepath? "lsua 2:Objects" or "lsua 2:Objects 3:MyNode". We just nee to test for = or : in name for guessing and/or add option -p for path
* -n (default namespace) then we could just do "lsua opc.tcp://localhost:4841 Objects"

General
* maybe we should also add this tools to setup.py and install them. calling them lsua and discoverua? (there are allready a bunch og lsusb and other tools on my pc) 
 
other ideas?

alkor, 2015-11-17:
Yes, I totally agree with your ideas! Here are some more:

lsua:

* expose other parameters of Browse (referenceTypeId, nodeClassMask, ...) to filter nodes by type.
* print node type and access rights, probably in style of `ls -l`

readua:

* support HistoryRead (or should this be a separate tool?)

oroulet, 2015-11-18:
maybe the commands should start with ua and not end. it increases a lot discoverability in shell with tab:
uals, uaread, uawrite, uasubscribe

oroulet, 2015-11-18:
started draft of implementation there
https://github.com/FreeOpcUa/python-opcua/commit/cee01ee6062230e99cb66189c4e4423e2f2575bd

oroulet, 2015-11-18:
added uals. missing some functionality compared to your tool @alkor , feel free to add them

oroulet, 2015-11-18:
well , I had some time in trains today and implemented most of it

closed 2015-12-18 by oroulet