FreeOpcUa/python-opcua #69: Server Breaks atexit

zerox1212, 2015-12-11:
I'm not sure if this is a bug, or I don't understand how the atexit module works.

If you register atexit functions in a program that uses the opc ua server, the atexit functions never get called. Perhaps it's actually an issue with asycnio?

oroulet, 2015-12-11:
http://www.scriptscoop.net/t/eaa2345e173f/windows-why-cant-i-catch-sigint-when-asyncio-event-loop-is-running.html

Ja it seems asyncio has an issue with atexit. It is supposed to be fixed
with python 3.5. Can you check?

zerox1212, 2015-12-15:
Interesting. I will try to test it this week and update this issue.

zerox1212, 2015-12-28:
I tested this behavior in Python 3.5 and the same problem occurs.

The only way atexit will be called is if server.stop() is called first. It certainly is strange because even
```python
raise SystemExit``` is ignored.

It isn't a huge problem, it just ruins my plan to dump a backup of the address space whenever the program exits. No I will have to make a more custom solution.

oroulet, 2015-12-28:
why not using  try: xxx finally: server.stop()?
This is what I use all the time and it is much cleaner than the atexit stuff anyway

closed 2015-12-28 by zerox1212