FreeOpcUa/python-opcua #71: Use of pycrypto

zerox1212, 2015-12-15:
Just grabbed the latest code and noticed https://github.com/FreeOpcUa/python-opcua/commit/dcbc5dda918c9ff1687e158d02d8d03d384fdf3f adds the dependency of pycrypto.

Unfortunately it looks like pycrpto is not supported by Python 3.4+.

Can we make the import of pycrypto only happen if we actually want to use certificates? Otherwise the server can't start with python 3.4 and newer.

oroulet, 2015-12-15:
1) yes i agree pycrypto should be optional
2) are you really sure it does not work for python3.4? It works even for
pypy.....

alkor, 2015-12-15:
Hmm, pycrypto seems to be in unhealthy state: https://github.com/dlitz/pycrypto/issues/158

oroulet, 2015-12-15:
Does someone has a better solution? state of crypto i python seems not very good

alkor, 2015-12-15:
Is https://github.com/pyca/pyopenssl an option? It's a wrapper around openssl library.

Some people recommend using https://github.com/pyca/cryptography , I'll try to read more about it.

oroulet, 2015-12-15:
you might have found something here @alkor pycripto is not really used as of today, only to sign certificate (works) and encrypt password (broken)

oroulet, 2015-12-15:
The description of cryptography module looks promising indeed! https://cryptography.io/en/latest/

zerox1212, 2015-12-15:
Maybe NaCI fits OPC UA better?
https://pynacl.readthedocs.org/en/latest/

oroulet, 2015-12-15:
just commited a try catch before uacrypto import so it is now possible to run without pycrypto... until we find a better solution

oroulet, 2015-12-15:
looks like cryptography has everything we need and a much better api than pycrypto, so the choice is quite obvious. It uses somce c code, but that seems to be very little

zerox1212, 2015-12-28:
Since cryptography looks like a good solution I'll close this issue.

closed 2015-12-28 by zerox1212