FreeOpcUa/python-opcua #84: Event subscription to .Net Server

chranagno, 2015-12-21:
Hi.
I have a  .Net Opc Ua server which generates events of SystemEventType.
I have managed to subscribe to the events using a .Net client.
However I cannot do the same using the python client (only the data subscription works).
On the other hand the python client works fine with the example python server.
I have used the subscribe_events function with no arguments and subscribe_events(2253,2130)
Am I doing something wrong?
Thanks

oroulet, 2015-12-21:
it should work but the event system is not very well tested so maybe you
found a bug.
What kind of events do you generate?
I just found å bug and commited a fix in dev2 branch maybe you can try it.

Otherwise you may send me an exe sending events and I can try ut

chranagno, 2015-12-21:
thanks for the immediate response
I tried the dev2 version with no success.
I want to generate events of a custom type but now I generate SystemType or BaseType events just for testing purposes. 
I'll try to send you an exe tomorrow

oroulet, 2016-01-05:
what is the status?

mertz3hack, 2016-03-07:
Is the script supposed to stay open when a data change subscription is started?
When I try to use this it only runs one time before disconnecting. 
There is about a 15second delay time before the data changes again.

oroulet, 2016-06-12:
no update for while, closing

closed 2016-06-12 by oroulet