FreeOpcUa/python-opcua #89: Timestamp Timezone

kevincolyar, 2015-12-24:
I'm trying to figure out what timezone is being used a server's variables.  There appears to be an hour offset but I'm not sure how to change it.

oroulet, 2015-12-24:
No idea.. It should be GMT. Do å grep for datetime in opc-ua dir, change
constructors to explicit gmt and send a patch. It would be great

kevincolyar, 2015-12-28:
According to the docs, `datetime.now()` returns the current local date and time.

https://docs.python.org/2/library/datetime.html#datetime.datetime.now

oroulet, 2015-12-28:
then find out our to return gmt and send a patch.

kevincolyar, 2015-12-28:
Working on it... found it might not be an issue.  May be related to issue https://github.com/FreeOpcUa/python-opcua/issues/40 instead.

kevincolyar, 2015-12-29:
Fixed, see https://github.com/FreeOpcUa/python-opcua/issues/40.

closed 2015-12-29 by kevincolyar