FreeOpcUa/python-opcua #103: 1.0 and 2.0 Release

oroulet, 2016-01-05:
Thanks to @alkor we now have support for encryption in both server and client. The OPC-UA implementation is now quasi complete. There are some shortcuts around, but even the proprietary stacks are not 100% perfect. I am waiting a bit to be sure we catch the most obvious bugs, then I propose to release a 1.0 version.

If someone knowns of  are missing big part of the protocol let me know

2.0 will then be released when we support xml protocol, but I currently have no use for it so it will wait for someone motivated. It should not be very hard though...

oroulet, 2016-01-05:
one thing which is missing though is the possibility to set user and passwords server side, not sure if it is a requirement.... not important for my work..

zerox1212, 2016-01-05:
Really great work over all. Thanks so much for your hard work.

My only issue at the moment might be the ability to remove nodes. This would especially be required for making an OPC UA server with a GUI and communications drivers. For example if you wanted to make a gateway like Kepware or TOP Server, etc.

destogl, 2016-01-05:
I agree with @zerox1212 . Removing nodes would be IMHO must-to-have feature for 1.0

Honestly I need it for my project using pyhton-opcua, but it is not in the main focus so it will take some time until I try to implement it...

oroulet, 2016-01-05:
@zerox1212 @destogl Really implementing delete_nodes and delete_references is a 20 minutes work on client. A bit more on server, but it should not be difficult, just look at how create_nodes is implemented

zerox1212, 2016-01-05:
I'll take a look at the server side and see if it's something I can figure out. I'm not really using the client, maybe @destogl can look at that part.

huazh, 2016-01-06:
As I'm digging the subscription race condition, I found it is a rather complex problem because of multi threads and locking.
So I have some suggestions before 1.0 release:
* remove binary_server.py
* implement binary_client in asyncio style, remove Keepalive thread
* review non thread safe code, submit them to asyncio thread by call_soon/call_later to remove the explicit lock. The barrier between asyncio thread and user thread may be put in InternalServer/InternalSession (as server) or BinaryClient (as client).
I think it will make a more robust library for 1.0 release

oroulet, 2016-01-06:
Won't the server code become much more complex if we move all calls to asyncio? 

I was afraid to depend too much on asyncio but when we first use it we can just continue..... The stupide thing is that the best thing with asyncio is co-routines but we cannot use them because of pypy and python2...

huazh, 2016-01-06:
The server code will not be more complex. We can use decorator to automatically submit most of InternalServer/InternalSession method to asyncio loop by call_soon(), then wait for the result. In  contrast, it will make code simpler as we can remove explicit locks to avoid subtle deadlock or race condition.

I agree the best thing of asyncio is co-routines. To support pypy and python2, we can just use trollius on python3 as trollius could be used on python3 and it's compatible with asyncio (with some limitations).

oroulet, 2016-01-06:
>> The server code will not be more complex.

Can you make an example? just migrate a few methods and show diff?

> To support pypy and python2, we can just use trollius on python3 as trollius could be used on python3 and it's compatible with asyncio (with some limitations).

There is already a few places where trollis is used if asyncio is not available but trollius is limited. SO I ended up only using callbacks and no coroutines

huazh, 2016-01-06:
What I mean is sticking to trollius even on Python3.4. Then the code will be running on whatever version of python.

oroulet, 2016-01-06:
>> What I mean is sticking to trollius even on Python3.4
Yes I understand. The current code uses asyncio events when possible and trollius if not. That is better I think and not complicated at all.

huazh, 2016-01-06:
It will make the code limit to callback style only, not coro

oroulet, 2016-01-06:
Great. share your code early. you do not need to fix everything. Show it early so you get comments

huazh, 2016-01-06:
A dirty sample code:
```
diff --git a/opcua/server/internal_server.py b/opcua/server/internal_server.py
index c866ff4..198b18c 100644
--- a/opcua/server/internal_server.py
+++ b/opcua/server/internal_server.py
@@ -147,6 +147,29 @@ class InternalServer(object):
     def create_session(self, name, user=User.Anonymous, external=False):
         return InternalSession(self, self.aspace, self.subscription_service, name, user=user, external=external)

+from concurrent.futures import Future
+from functools import partial
+
+
+def async_and_wait(func):
+    def async_func(self, fut, *args, **kargs):
+        try:
+            rs = func(self, *args, **kargs)
+        except Exception as e:
+            fut.set_exception(e)
+        fut.set_result(rs)
+
+    def wrapper(self, *args, **kargs):
+        if self.external is False and self.loop.is_alive():
+            fut = Future()
+            afunc = partial(async_func, self, fut, *args, **kargs)
+            self.loop.call_soon(afunc)
+            return fut.result()
+        else:
+            return func(self, *args, **kargs)
+
+    return wrapper
+

 class InternalSession(object):
     _counter = 10
@@ -155,6 +178,7 @@ class InternalSession(object):
     def __init__(self, internal_server, aspace, submgr, name, user=User.Anonymous, external=False):
         self.logger = logging.getLogger(__name__)
         self.iserver = internal_server
+        self.loop = internal_server.loop
         self.external = external  # define if session is external, we need to copy some objects if it is internal
         self.aspace = aspace
         self.subscription_service = submgr
@@ -211,6 +235,7 @@ class InternalSession(object):
                 self.user = User.Admin
         return result

+    @async_and_wait
     def read(self, params):
         return self.iserver.attribute_service.read(params)
```

oroulet, 2016-01-06:
looks good for me!

oroulet, 2016-01-06:
one thing to think of  is to integrate with app that that have their own asyncio loop and would like that freeopcua uses their loop... but maybe it will be later...

huazh, 2016-01-06:
I will try to implement client on asyncio, then find the best place to set barrier between async thread and user thread

oroulet, 2016-01-12:
I looked again at you server decorator, and I think this is the only thing we need currently on server side .just call call_soon_threadsafe instead of call_soon so user can use wherever they want. Can you send a patch or do you think there are drawbacks? We can then try to remove locks.... and remove binary_server.py because it is then not usable anymore

huazh, 2016-01-12:
call_soon_threadsafe() is enough.
But I'm thinking about change user interface code (Server/Client/Node/Subscription) to async loop aware. It has advantage as follow:
1. User can supply an event loop to Server/Client, then the whole package of opcua can run in single thread.
2. If event loop is not supplied, Server/Client can create its own event loop thread, then provide a synchronized API to user. 
3. Server is using Client to register itself to discovery server currently. But Client provides synchronized interface only for now. It will block the server event loop thread. If the Server and Client can share the event loop, the problem can be fixed more easier.
4. No more subscription race condition. It will be much easier for user to handle subscription callback if user chooses to use asynchronous interface (the whole application running in a single thread)
5. Locks can be removed thoroughly.

zerox1212, 2016-01-13:
@destogl , can you check to see if this works?
https://github.com/FreeOpcUa/python-opcua/commit/4126a0284a866ba25dcf45b6c5863c2e652ffee9

I looked at the server, but I don't have time to implement it yet. Seems much more complicated.

oroulet, 2016-01-13:
@huazh OK but I would prefer that you do things progressively. Merging the decorator is easy. It should also be possible to do 1, 2 and 3 independently. big patches are hard to merge.

oroulet, 2016-01-13:
> If event loop is not supplied, Server/Client can create its own event loop thread, then provide a synchronized API to user. 

Do you mean creating an api that can be used both async through asyncio and standart sync call? that would be great but we should make it in client first, and we have not found a solution yet

destogl, 2016-01-14:
@zerox1212 it will try it on the weekend.

destogl, 2016-01-16:
@zerox1212 I managed to implement deleting of node on the server. It works but there is some issues when the subscribed node is deleted. Detailed description of problem in #114

zerox1212, 2016-01-18:
Nice work @destogl . I'm glad you got time to do it, it looks like I probably didn't have the skills to implement it when I started digging into the server.

@oroulet we can update the readme that deleting nodes is now implemented.

destogl, 2016-02-10:
Hi all,

what should be done to get to 1.0. In the last weeks there was a lot of smaller and bigger bug fixes/improvements. Also deleting of node is now implemented...

@oroulet I propose to create a list of actions  needed to get to 1.0....

oroulet, 2016-02-10:
> I propose to create a list of actions needed to get to 1.0....
yes the milestone is already created so we can just flag bugs/enhancement we want.
Not much left in my opinion. 

@huazh wants a rewrite to async, but it is not as easy he thought so maybe we can just leave it for now and work on it for 2.0, at that time we may just drop support for python2.7 and things will get easier. 

Removing server locks should be easy, we just need to push InternalSession calls to asyncio loop. Maybe it is something to try? @huazh do you see any issue with this? I can just implement your solution from above

I think we need some tests for encryption for 1.0. This is something I will do.

Anything else?

history might be coming but we can flag it to 1.1

destogl, 2016-04-11:
So do we wait async or we want to mark it as 1.0?

It would help to make todo list if there is some things to do.
IMO I think this is finished. Asyncio could wait, since there is no much development currently on this line...

oroulet, 2016-04-11:
* async on client side won't be implemented for 1.0. Trollius is dead, this means that an async solution will only work on python3.
* On server side replacing lock with pushing all request to asyncio loop is tempting. Locking is always a risk for hanging.
* Then it looks like we still have an issue in client with long lasting connections:
https://github.com/FreeOpcUa/python-opcua/issues/152
and 
https://github.com/FreeOpcUa/python-opcua/issues/157
* Then your event stuff should be merged

Anyway I do not see version numbering as very important, python-opcua seems to be very stable for many users

destogl, 2016-04-11:
OK. I do it in nicer format:

- [ ] Server side: Replace locks with pushing all request to asyncio loop
- [ ] Issue with client long lasting connection is solved #152 and #157 
- [ ] Event implementation (#133) is finished

Yes I agree, but it would be nice to have 1.0 and maybe close this issue and open new one for 2.0