FreeOpcUa/python-opcua #106: Detecting connection loss

bitmarker, 2016-01-06:
Hello!
Is it possible to detect if the server has been turned off or the connection between the server and client was lost? The only way I found is to subscribe to the server time and use the subscription as "watchdog signal". Is there any way to detect if the session has ended?

alkor, 2016-01-06:
UaExpert client periodically reads state of server (node i=2259). Probably that requires less network traffic than subscription.

alkor, 2016-01-06:
Found in OPC UA specs, part 4, 6.5:

> The preferred mechanism for a Client to monitor the connection status is through the keep-alive of the Subscription. A Client should subscribe for the State Variable in the ServerStatus to detect shutdown or other failure states. If no Subscription is created or the Server does not support Subscriptions, the connection can be monitored by periodically reading the State Variable.

oroulet, 2016-01-07:
Interesting to see than that kind of thing is in spec. Yes it seems to be
the most reliable way to detect connection issues. Many sdk have this
built-in, and we may implement it but it requires to remember a bunch of
state things ( like running subscriptions). We wait anyway for the port of
client to asyncio....

zerox1212, 2016-01-07:
Maybe this is related. Is it possible to override the server state to force the client to disconnect on it's end? I'm still looking for ways to force all the clients to disconnect from my server.

ronytesler, 2021-11-25:
"A Client should subscribe for
the State Variable in the ServerStatus to detect shutdown or other failure
states."
OK, but what should we do with the value we get?!
I got 4 as a value, what does that mean?

AndreasHeine, 2021-11-25:
https://reference.opcfoundation.org/v104/Core/DataTypes/ServerState/

alexrudd2, 2024-02-23:
Dead link.  It's now https://reference.opcfoundation.org/Core/Part5/v104/docs/12.6