FreeOpcUa/python-opcua #111: missing generic ExtensionObject class

oroulet, 2016-01-14:
def extensionobject_from_binary(data):
 ....
    elif TypeId.Identifier not in ExtensionClasses:
        raise UAError("unknown ExtensionObject Type: {}".format(TypeId))
,.,,,

This is wrong, server can create custom extension objects. Can you have a look @alkor ? you wrote that code and I am unsure to fix it cleanly.
We just need to recreate the base ExtensionObject class and return it with binary or xml body

try this to see error:
python tools/uaread --url opc.tcp://opcua.demo-this.com:51210/UA/SampleServer --path 0:Objects,2:Data,2:Static,2:Scalar,2:StructureValue

alkor, 2016-01-14:
Oh, I thought that there's no need for explicit ExtensionObject. I'll try to fix this ASAP.

oroulet, 2016-01-14:
Anyway you did a very good job with this extension stuff! It was the
ugliest thing in the stack

oroulet, 2016-01-16:
also this one fails:
python tools/uaread --url opc.tcp://opcua.demo-this.com:51210/UA/SampleServer --path 0:Objects,2:Data,2:Static,2:Array,2:VariantValue

This seems to be something a bit different, some NodeId types are missing in the autogenerated ExtensionObject dict

closed 2016-01-19 by oroulet