FreeOpcUa/python-opcua #112: Cannot read and write a whole matrix ( only single value )

hitokiri8x, 2016-01-15:
Hello, 
with the client it's not possibile to read and write a matrix of data as a whole item, it's only possible accessing the single value of the matrix.
The exception it the following:
```
Traceback (most recent call last):
  File "C:\Users\alessandro\git\HMI-web\initMLC.py", line 59, in <module>
    writeDouble(tagTest2,singolo)
  File "C:\Users\alessandro\git\HMI-web\initMLC.py", line 22, in writeDouble
    te= tag.get_data_value()
  File "C:\Python34\lib\site-packages\opcua\common\node.py", line 177, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "C:\Python34\lib\site-packages\opcua\common\node.py", line 242, in get_attribute
    result = self.server.read(params)
  File "C:\Python34\lib\site-packages\opcua\client\binary_client.py", line 266, in read
    response = ua.ReadResponse.from_binary(data)
  File "C:\Python34\lib\site-packages\opcua\ua\uaprotocol_auto.py", line 7948, in from_binary
    return ReadResponse(data)
  File "C:\Python34\lib\site-packages\opcua\ua\uaprotocol_auto.py", line 7925, in __init__
    self._binary_init(binary)
  File "C:\Python34\lib\site-packages\opcua\ua\uaprotocol_auto.py", line 7957, in _binary_init
    array.append(DataValue.from_binary(data))
  File "C:\Python34\lib\site-packages\opcua\ua\uatypes.py", line 868, in from_binary
    value = Variant.from_binary(data)
  File "C:\Python34\lib\site-packages\opcua\ua\uatypes.py", line 789, in from_binary
    vtype = VariantType(encoding & 0b01111111)
  File "C:\Python34\lib\enum.py", line 222, in __call__
    return cls.__new__(cls, value)
  File "C:\Python34\lib\enum.py", line 457, in __new__
    raise ValueError("%r is not a valid %s" % (value, cls.__name__))
ValueError: 75 is not a valid VariantType
```

d09971c  in the new brancharrays is a commit to attempt to fix this "bug".
Right now seems to work, will keep to test and work to see if it broke something else.

oroulet, 2016-01-15:
Can you try to write?

hitokiri8x, 2016-01-15:
I have re-tested now ( a quick test was done when I posted the issue ) : 
Write of single value: OK
write of a value of an array: OK
Write of a complete array: OK
write of single value of a matrix: OK
write of a complete matrix : OK

oroulet, 2016-01-16:
I have done somw more work on this, now we return lists of lists. This is more compatible with rest of API and even easier to convert to and from numpy. Does it work in your case? No invertions?

oroulet, 2016-01-16:
https://github.com/FreeOpcUa/python-opcua/tree/arrays

oroulet, 2016-01-16:
Can you write a very simple minimal example file reading and writting arrays and converting to numpy? I have not server to test this. Thanks

hitokiri8x, 2016-01-18:
Thanks, I've tested now and it all works, both passing a single dimension list of all value and with a list of list.
No inversion are done, the value are parsed to and from the plc in the correct order.

Right now I'm not using numpy, I'm evaluating the option right now, in any case in the following days I'll write some example with numpy!

oroulet, 2016-01-19:
OK let me know if you find issues with this

closed 2016-01-19 by oroulet