FreeOpcUa/python-opcua #118: add support for views in server

oroulet, 2016-01-18:
not planned currently

maljac, 2016-08-01:
Hi @oroulet 

just wanted to know: what's the reason for not supporting views.
Does it need a fundamental change in this project or is it only related to missing time and focussing on other features.

oroulet, 2016-08-01:
I haven't had need for it yet. I have not read how they should be used either.....

maljac, 2016-08-01:
I havent seen them in action so far, but as far as I understand the concept, you can sort of compose your own view. You can imagine this like a SQL view. 

Example: Show all nodes of type X that have a certain attribute (like the physical location). Thus you have all your machines that are located in a specific location on own view.

zerox1212, 2016-08-15:
My understanding is that it is meant to be used with users.

For example, is a normal user connects to the server, they only see views with read only data. If an super user connects to the server, maybe he has a different view with parameters, etc.

mikakaraila, 2018-06-09:
I am using views for process area and IO topology with node-opcua.

oroulet, 2018-06-09:
@mikakaraila what are views? Who is supposed to make them? Are they jus links to existing nodes in address space?

mikakaraila, 2018-06-09:
Yes you can build hierarchy and then use reference that points to existing node in address space.
I am using server with method that loads file that is usea to create view.

oroulet, 2018-06-09:
So the conclusion is that we do not need any new functionality for using views?