FreeOpcUa/python-opcua #121: datetime decoding fails

oroulet, 2016-01-22:
I sometims get such a crash:
    return ReadResponse(data)
  File "/home/olivier/python-opcua/tools/../opcua/ua/uaprotocol_auto.py", line 7853, in __init__
    self._binary_init(binary)
  File "/home/olivier/python-opcua/tools/../opcua/ua/uaprotocol_auto.py", line 7885, in _binary_init
    array.append(DataValue.from_binary(data))
  File "/home/olivier/python-opcua/tools/../opcua/ua/uatypes.py", line 1046, in from_binary
    obj.SourceTimestamp = unpack_datetime(data)  # DateTime.from_binary(data)
  File "/home/olivier/python-opcua/tools/../opcua/ua/uatypes.py", line 199, in unpack_datetime
    return win_epoch_to_datetime(epch)
  File "/home/olivier/python-opcua/tools/../opcua/ua/uatypes.py", line 55, in win_epoch_to_datetime
    return FILETIME_EPOCH_AS_DATETIME + timedelta(microseconds=epch // 10)
OverflowError: date value out of range
INFO: Socket has closed connection

@alkor Do you have any idea what is happening? you wrote that code
It happens sometimes while reading

 python tools/uaread --url opc.tcp://opcua.demo-this.com:51210/UA/SampleServer --path 0:Objects,2:Data,2:Dynamic,2:UserArray,2:VariantValue -v DEBUG --timeout 4

you may need commit https://github.com/FreeOpcUa/python-opcua/commit/9875f0d6cf7a5005587d4ee272ed56f911e3dd94 if you want to test

alkor, 2016-01-22:
Fixed this issue, but reading this value still fails sometimes. Now I'm getting some crazy string sizes like this:

```
opcua.common.utils.NotEnoughData: Not enough data left in buffer, request for 1006632963, we have Buffer(size:1826, ...)
```

oroulet, 2016-01-23:
Thanks. I can look at the next one

closed 2016-01-23 by oroulet

oroulet, 2016-01-24:
Looks like server sends bad packets. Wireshsrk cannot parse them either

PR #123 (2016-01-23): Catch overflows in win_epoch_to_datetime()
Fix #121: FILETIMEs after 31 Dec 9999 can't be converted to datetime.