FreeOpcUa/python-opcua #125: Allow users to change the password encryption algorithm

sscherfke, 2016-01-25:
Currently, the high-level client `opcua.Client` uses the algorithm *rsa-oaep* to encrypt session passwords. This is hardcoded in lines [376](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/client/client.py#L367) and [370](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/client/client.py#L370).

However, some servers seem to not suport it but rather require, for example, *rsa-1_5*. In order to support this I have to change `opcua/client/client.py:367+` as follows:

```python
            # data = uacrypto.encrypt_rsa_oaep(pubkey, etoken)
            data = uacrypto.encrypt_rsa15(pubkey, etoken)
            params.UserIdentityToken.Password = data
        params.UserIdentityToken.PolicyId = self.server_policy_id(ua.UserTokenType.UserName, b"username_basic256")
        # params.UserIdentityToken.EncryptionAlgorithm = 'http://www.w3.org/2001/04/xmlenc#rsa-oaep'
        params.UserIdentityToken.EncryptionAlgorithm = 'http://www.w3.org/2001/04/xmlenc#rsa-1_5'
```
A method like `Client.set_pwd_encryption_algorithm(algorithm_string)`, which takes strings like `'rsa-oaep'`, `'rsa-1_5'`, or any other valid algorithm names, would solve this issue.

alkor, 2016-01-25:
If I understand specs correctly, client must read SecurityPolicyURI from the endpoint description, and then encrypt password with respective asymmetric algorithm. If the endpoint uses Basic128Rsa15 policy, then encrypt with rsa-1_5. If Basic256, then rsa-oaep.

I'll try to implement and test this thing.

oroulet, 2016-01-25:
Sounds logic

destogl, 2016-02-09:
Is this solved by #128?

oroulet, 2016-02-09:
I thjink so but I also remember that someone said there were still an
issue... do not remember where I read it

sscherfke, 2016-02-15:
Works for me. Thx :)

destogl, 2016-02-16:
@sscherfke please close the issue than :)

closed 2016-02-16 by sscherfke