FreeOpcUa/python-opcua #126: Create folder/variable with node id like ns=2;s="foo.bar"

kevincolyar, 2016-01-30:
Is this possible?  It appears add_variable/add_folder only takes a namespace and name which generates node id's like ns=2;i=1234

oroulet, 2016-01-30:
do you know any other SDK that accept quotation inside quotation?  ns=2;s=foo.bar should work.

kevincolyar, 2016-01-31:
Sorry, the quotes aren't needed.

I tried passing ns=2;s=foo.bar to add_variable but it just made that it's name.  The node id was stil ns=2;i=1234

Do I need to pass it a node instance?

oroulet, 2016-01-31:
Mynode.add_variable(nodeid, browsename) where args are either strings or
NodeId and QualifiedName

kevincolyar, 2016-02-01:
Thanks @oroulet!

For reference, I was able to do:

```python
node.add_variable("ns=2;s=foo.bar", "bar", value)

closed 2016-02-01 by kevincolyar