FreeOpcUa/python-opcua #135: Attempt to connect to Arburg proprietary server

mertz3hack, 2016-02-17:
Arburg is an injection molding machine company that runs OPC UA servers on their machines.
I am able to connect to the machine and browse with UAExpert.

Taking any suggestions.  Can I set the URI?

I am trying to connect to the server with

<code>import sys
sys.path.insert(0, "..")
from opcua import Client

client = Client("opc.tcp://admin:password@192.168.12.28:4880/Arburg")
client.connect()
client.disconnect()</code>

I get this when I run it

<code>
Traceback (most recent call last):
  
File "minimal-client.py", line 11, in <module>
    client.connect()
  
File "..\opcua\client\client.py", line 208, in connect
    self.activate_session(username=self.server_url.username, password=self.server_url.password, certificate=self.user_ce
rtificate)
  
File "..\opcua\client\client.py", line 384, in activate_session
    self.server_policy_uri(ua.UserTokenType.UserName))
  
File "..\opcua\crypto\security_policies.py", line 451, in encrypt_asymmetric
    raise UaError("Unsupported security policy `{}`".format(uri))
NameError: name 'uri' is not defined
ServiceFault from server received  in response to ReadRequest
Exception in thread Thread-2:

Traceback (most recent call last):
  
File "C:\python\lib\threading.py", line 914, in _bootstrap_inner
    self.run()
  
File "..\opcua\client\client.py", line 51, in run
    val = server_state.get_value()
  
File "..\opcua\common\node.py", line 81, in get_value
    result = self.get_data_value()
  
File "..\opcua\common\node.py", line 90, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
 
 File "..\opcua\common\node.py", line 155, in get_attribute
    result = self.server.read(params)
  
File "..\opcua\client\ua_client.py", line 267, in read
    data = self._uasocket.send_request(request)
 
 File "..\opcua\client\ua_client.py", line 75, in send_request
    self.check_answer(data, " in response to " + request.__class__.__name__)
  
File "..\opcua\client\ua_client.py", line 84, in check_answer
    hdr.ServiceResult.check()
 
 File "..\opcua\ua\uatypes.py", line 323, in check
    raise UaStatusCodeError("{}({})".format(self.doc, self.name))
opcua.common.uaerrors.UaStatusCodeError: The session cannot be used because ActivateSession has not been called.(BadSessionNotActivated)
</code>

oroulet, 2016-02-17:
looks like  bug in exception handling here. I will commit a fix, the encryption stuff is missing tests and that king of things happen....
I see we are missing an example client showing how to use encryption.
first call uadiscovery opc.tcp://yourserver to see what encryption your server supports then
call set_security_string() method in client to set correct encryption mode

joaopmrod, 2016-02-18:
I'm experiencing exactly the same problem with Ignition Software's OPC-UA Server.

oroulet, 2016-02-18:
Can one of you put their server on internet or record a session with wireshark so we understand what is happening.

mertz3hack, 2016-02-18:
@oroulet I am not able to put the server on the internet and wireshark is not recognizing my wireless card via windows on a surface book :/
when I connect with UaExpert it says the security policy is 'http://opcfoundation.org/UA/SecurityPolicy#None' and the ApplicationUri is 'uri://Arburg/App/Selogica/direct'

oroulet, 2016-02-18:
Then set logging to debug and send terminal output. Maybe we'll see
something there

mertz3hack, 2016-02-18:
@oroulet I downgraded to version 1.12.9 of wireshark and it is now seeing my wireless card. 
the ip address of 192.168.12.27 is the server I am trying to connect too.
Is this what you were looking for?
[machineConnect.zip](https://github.com/FreeOpcUa/python-opcua/files/137033/machineConnect.zip)

oroulet, 2016-02-19:
the log seems corrupt. I cannot see the request freeopcua sends. But from the responses we see, the connection has succeeded and you read some nodes values then tried to subscribe to a node and that is where something fails. I cannot see what node you are trying to subscribe too, but the server sends a serviceFault response. This looks very different to what you say and the error message you wrote above... How does you code looks like? does it fail even you do not attempt to subcsribe a node?  You can only subscribe to variable nodes

mertz3hack, 2016-02-19:
I'm not exactly sure what I should be looking for
The attached zip shows the initial part of the error using the minimal-client without doing uadiscovery.
There is also a successful connection with UAexpert
[output.zip](https://github.com/FreeOpcUa/python-opcua/files/138437/output.zip)

oroulet, 2016-02-20:
Can you fix the typo in line 451 in security_policies.py (uri -> policy_uri) then run again?
or fetch pull request #136 it has the fix

It will not fix your issue but at least we will get a good error message

oroulet, 2016-02-20:
@alkor In this log I see that in UserTokenPolicy the PolicyId is set to UserName_Policy_1. Is this a normal PolicyId? Can it be the reason of the crash here? any suggestion t fix this? 
Then SecurityPolicyUri for UserTokenPolicy is http://opcfoundation.org/UA/SecurityPolicy#None. Maybe that means that password should not be encrypted?

oroulet, 2016-02-20:
@alkor Am I right to understand that name og UserTokenPolicy does not mean anything? The important thing is what is written in the SecurityPolicyUri? and in that case we are missing support for the simplest one: no encryption?

alkor, 2016-02-20:
Oh, yes, I see the problem. I'll fix this. Thank you for fixing my typo (uri -> policy_uri).

oroulet, 2016-02-20:
This should now be fixed in master. Can you test and report?

mertz3hack, 2016-02-22:
This is what I tried running to connect
<code>
import sys
sys.path.insert(0, "..")
from opcua import Client
if __name__ == "__main__":
    client = Client("opc.tcp://admin:password@192.168.12.27:4880/Arburg")
    try:
        client.connect()
    finally:
        client.disconnect()
</code>

And the results
<code>
Sending plain-text password
Exception in thread Thread-1:
Traceback (most recent call last):
  File "C:\python\lib\threading.py", line 914, in _bootstrap_inner
    self.run()
  File "C:\python\lib\threading.py", line 862, in run
    self._target(*self._args, **self._kwargs)
  File "..\opcua\client\ua_client.py", line 92, in _run
    self._receive()
  File "..\opcua\client\ua_client.py", line 99, in _receive
    msg = self._connection.receive_from_socket(self._socket)
  File "..\opcua\ua\uaprotocol_hand.py", line 642, in receive_from_socket
    header = Header.from_string(socket)
  File "..\opcua\ua\uaprotocol_hand.py", line 121, in from_string
    hdr.MessageType, hdr.ChunkType, hdr.packet_size = struct.unpack("<3scI", data.read
  File "..\opcua\common\utils.py", line 101, in read
    chunk = self.socket.recv(size)
OSError: [WinError 10038] An operation was attempted on something that is not a socket
</code>

oroulet, 2016-02-22:
Set logging to debug and run  wireshark again. Then copy paste stdout here. And please try to keep some formating when pasting.
And add a sleep after connect to make sure we do not close connection too early

oroulet, 2016-02-22:
please update master. I just added some more debug to client. Now it prints everything it sends and receives from server, so you do not need to use wireshark.
so set logging to debug, run your example and paste output here

mertz3hack, 2016-02-22:
@oroulet is there an instant messenger on here we can use?

mertz3hack, 2016-02-22:
I am able to **successfully** connect too and specify a node to get it's value.
I am still getting an error that I am not sure about yet.

[minimal-client.txt](https://github.com/FreeOpcUa/python-opcua/files/141506/minimal-client.txt)
[minimal error.txt](https://github.com/FreeOpcUa/python-opcua/files/141507/minimal.error.txt)
[minimal debug.txt](https://github.com/FreeOpcUa/python-opcua/files/141511/minimal.debug.txt)

oroulet, 2016-02-23:
the error is not visible in the minima debug.txt file. Should I understand that i only happens sometime? and only at end of connection?
Anyway it is probably some windows specific things at closing. Can you open a bug request for it? and close this one?
Thanks

mertz3hack, 2016-02-23:
I will close this issue with successful connection and will see if I can find anything more about his other error I am getting before creating an Issue for it.

closed 2016-02-23 by mertz3hack