FreeOpcUa/python-opcua #137: Calling server.stop() doesn't clean up connections.

zerox1212, 2016-02-19:
Calling stop() on the OPC UA server doesn't close the asyncio transport object and clean up active connections. This causes the server to crash if you call the "dump address space" method while clients are connected, even after the server is stopped.

Python documents say that you should close any transports associated with asyncio explicitly.
https://docs.python.org/3/library/asyncio-dev.html#close-transports-and-event-loops

Not a major issue, but is annoying for anyone who wants pickle the address space to a file. I tried to fix it but wasn't able to do it correctly.

oroulet, 2016-02-19:
Can you fix the typo in line 451 in security_policies.py (uri -> policy_uri) then run again?
or fetch pull request https://github.com/FreeOpcUa/python-opcua/pull/136 it has the fix

It will not fix your issue but at least we will get a good error message

oroulet, 2016-02-20:
I had a look yesterday. I think I understand what is happening here. The asyncio transport does not close as long as some clients are still connected, but I have no idea how to fix this ... I am keeping this in mind..

zerox1212, 2016-02-24:
Why can't we call transport.close() in the Binary Server's stop() method? This is what I tried to do in my fork, but it broke tests.

oroulet, 2016-02-25:
I am not sure I understand everything but it looks like a new transport is created for every client. and these transports are not closed when calling close on the the BinaryServer class. This is a good case to ask on stackexchange.... not idea how to solve that

zerox1212, 2016-03-08:
Made a post on stackoverflow.
http://stackoverflow.com/questions/35872750/how-to-close-python-asyncio-transport

oroulet, 2016-03-08:
Great!
You made a good description of the issue but you risk to get complains that
the code sample is too long. You should try to make a simpler example.

zerox1212, 2016-03-08:
I know the sample is long, but I don't know how to capture the core of the problem with less code. If you can make a snippet I will edit the post.

zerox1212, 2016-03-08:
Take a look at https://github.com/FreeOpcUa/python-opcua/pull/143 please.

destogl, 2016-03-10:
Is this solved with #143? Are some additional test needed?

zerox1212, 2016-03-10:
If you can start the server, have 2 or more clients connect, then disconnect from the client side and make sure the correct transport object is removed from the list it would be great. This is the only test I didn't try.

zerox1212, 2016-03-17:
Have not heard of any problems so far. I'll close this issue.

closed 2016-03-17 by zerox1212