FreeOpcUa/python-opcua #141: client.disconnect error

mertz3hack, 2016-03-01:
I have been getting the following error when disconnecting that I have not been able to figure out.
[clientdisconnecterror.txt](https://github.com/FreeOpcUa/python-opcua/files/153577/clientdisconnecterror.txt)

I am connecting with the following file.
[client.py.txt](https://github.com/FreeOpcUa/python-opcua/files/153581/client.py.txt)

This file is with Logging set to Debug.
[clientdebug.txt](https://github.com/FreeOpcUa/python-opcua/files/153579/clientdebug.txt)

Any thoughts on getting rid of this error?

oroulet, 2016-03-02:
Seems to be a Windows specific error when cancelling a read operation. I
can add a try/catch there but this looks like an error without consequence

mertz3hack, 2016-03-02:
@oroulet I have gotten the same error with new python installations of 3.4.0, 3.4.4, and 3.5.1

oroulet, 2016-03-04:
Yes you do,  but I still think this is not really an important issue

mertz3hack, 2016-03-09:
@oroulet 
So this issue wasn't affecting me when I was running a python script but I tried to go one step farther using py2exe
The exe would always show a disconnect error pop up after the exe is disconnected and closed
My current solution to avoid the error (which is not a good solution) is to just not disconnect since the server will eventually kick the user for inactivity

oroulet, 2016-03-09:
ok let's try to catch it...

mertz3hack, 2016-03-16:
@Any suggestion on where/how to track this?

oroulet, 2016-03-16:
Well the stack trace tells everything:
le "..\opcua\common\utils.py", line 101

Add oserror to the try catch and send a pull request

mertz3hack, 2016-03-28:
@oroulet sorry nOOb here.  I need a little more information on what you want me to do

oroulet, 2016-03-28:
I implemented it already. The last version on pip should have the fix. Do you still get the error?

mertz3hack, 2016-03-28:
@oroulet seems to be working! Thanks now I will just have to track back the change log to see what you did.  Do you know which commit it was solved in?  I have been MIA for a bit and haven't watched a bunch.

I also didn't know this project was on pip!

closed 2016-03-28 by mertz3hack

mertz3hack, 2016-03-28:
https://github.com/FreeOpcUa/python-opcua/commit/b88ad958ea3b240dfa5a51aa13c8347ea572c783