FreeOpcUa/python-opcua #142: embedding OPC UA client into Django to build web SCADA

alamsha, 2016-03-04:
Is it possible to embed the python-opcua client into Django or any other CMS framework and build dynamc web SCADA pages?.

oroulet, 2016-03-04:
if django let you create python threads it will work. but I have no idea if it does

alamsha, 2016-03-05:
Looking for some guidance and sample code to start building dynamic web pages with OPC UA client.

oroulet, 2016-03-05:
Ask the django people. They have an entire framework, freeopcua is just a library

zerox1212, 2016-03-07:
Making an OPC UA scada with Django is going to be a big project. Like @oroulet said, you need to take a look at the Django tutorials. If you are trying to do something simple it might be better to use WSGI or flask or something. Maybe CherryPy would be a start.

closed 2016-04-01 by oroulet

joaolrc, 2022-12-23:
Hello

Django dev here.
Maybe a bit late to answer in this thread but my team has the current need to implement an event driven OPC-UA interface on an existing Django project with a reasonably large codebase. 

I would like to know if anyone has implemented it already and if so, whats the best way to do it. 
I was thinking of implementing it on top of asyncio using this https://github.com/FreeOpcUa/opcua-asyncio 
 (we have done some things with this approach already and seem to work well)

Any comments and ideas are deeply appreciated.
Thanks in advance.

oroulet, 2022-12-29:
Hi @joaolrc ,
Original opcua-asyncio dev here. I do not know django so I cannot say much about the best way to expoe an UA interface from django.
If you want to expose data then the best way to do it currently is to react to internal django events and fill up the UA node with the values you want at that point using `write_attrtibute_value`method. Depending on your architecture it might result in some duplicated data since the value is also stored in the UA server.
There was some work originally to directly query some underlying database (grep for value_callback) but I think nobody has ever used that and it required to manually  trigger UA events everytime your django db changes and that API might be missing

oroulet, 2022-12-29:
if you only want to fire events, then the easiest is to build your events and fire them when you want. That should be easy.

Also just so I have said it: I wrote that software to fill up a need. security was not part of my need, although there have been many patches and improvments on that side over time

oroulet, 2022-12-29:
and you should really discuss these things in a discussion thread instead of hijacking a closed issue!

joaolrc, 2022-12-29:
Hi @oroulet 

Thanks for your reply.

The current requisites are just for writing some variables in a OPC-UA server when some django events happen, which will then be read by robot(s) to move stuff around. 
I am still introducing myself to this protocol but  your advice seems to make perfect sense. 

I believe that, for this feature, security is not a big concern as this will work on LAN or through VPN. 
But later, things may change and we are thinking about deploying stuff in cloud infrastructures, which, at that point, things may have to be reviewed.

Regarding posting this here, i thought that it was convenient as this was related to the original post. But i get it, django forums may apply better for this kind of stuff.

Kind Regards!