FreeOpcUa/python-opcua #144: Server UserTokenPolicy of type UserName disabled?

destogl, 2016-03-10:
Hi all,

I just tried to connect as admin to the server and since I had some issues I saw that TokenPolicy of type UserName is not in the list of UserIdentityTokens (Look this [line](https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/server.py#L231)).

I think here is missing `idtoken4` so the line should be:

`edp.UserIdentityTokens = [idtoken, idtoken2, idtoken3, idtoken4]`

oroulet, 2016-03-10:
We have an issue here. We need to let user enable or disable enpoints

destogl, 2016-11-15:
This is kind a solved because idtoken4 is in the list. The question is only should we let user enable/disable IdentityTokens?

oroulet, 2016-11-15:
When setting up a server we should definitely let users decice what login methods are available.