FreeOpcUa/python-opcua #145: Updating Values Without Triggering Server Subscriptions

kevincolyar, 2016-03-15:
I currently have a handler that is called via a subscription when my values are changes by a connected client.  I'm looking for a way to update the same values from the server **without** triggering the subscription.  The subscriptions are what tell me a client action has modified a value, I don't want those to trigger when they are modified server side.

Ideally I'd like to only subscribe to data changes coming from clients.  Ideas?

oroulet, 2016-03-16:
You need to modify the db dict directly if you want such a thing. But why? Do you have a use case?

kevincolyar, 2016-03-18:
Is it possible to subscribe to client write events?  That would probably be a better solution if it's possible.

zerox1212, 2016-03-21:
I don't really understand why you would want to do that.
Wouldn't it be better to have two separate nodes?

Node 1 is read only that your server software writes to.
Node 2 is a normal client subscribed node.

Then your application can  update the server side node whenever a client writes to node 2.

oroulet, 2016-03-21:
No. Not possible but as Andrew asked: why would you want such a thing?

kevincolyar, 2016-03-23:
Closing this. I found a better way of going about what I was trying to solve.

closed 2016-03-23 by kevincolyar