FreeOpcUa/python-opcua #146: Automatic Recconnect

mertz3hack, 2016-03-18:


destogl, 2016-03-23:
There was few time discussion about that and it was decided that the highest-level client implementation should take care about that... This you can do by catching the exceptions.

@oroulet Am I right?

KeepAlive should take care that your connection doesn't time out, so the question is where is really your problem. Maybe the server has short time-out...

oroulet, 2016-03-23:
Automatic reconnect requires quite a lot of code and internal changes.
Since the client would then also need to remember subscriptions and attempt
to reconnect to session, etc,. So I am unsure how to implement this

Rosepeter, 2017-11-23:
What about adding some kind of callback-interface to the client that informs the application when e.g. timeout exception in the KeepAlive-thread. At the moment the only way to get informed about this, is using the synchronous interface of Client and this is often is not the best choice (e.g. when calling a method).
When you start to use the underlying UaClient or even the UASocketClient with callbacks it seems not to be possible to get informed about timeouts in the KeepAlive-thread. Of course, you may claim that your on your own, when using that stuff. But it would be very helpful to use Client for setting up the connection etc. and then use these low-level interfaces to establish asynchronous communication with the server.
If anybody is interested in such a callback-interface I could provide a basic-implementation.