FreeOpcUa/python-opcua #151: Improve node method get_data_type

zerox1212, 2016-03-28:
The way it is right now ```node.get_data_type``` only returns the xNodeID with ObjectID. You then have to look up the OPC UA spec or go into object_ids.py and decode it.

Is there a reason the high level method can't also return a string of the data type?

oroulet, 2016-03-28:
Good point. We should auto generate reverse objectids. You can open bug
request and even better implement it look at py files under schema directory

oroulet, 2016-07-26:
ObjectIs cannot be enum has this is has huge performance impact. The reversed objectId dict is now created but it would be strange to return a string instead of a nodeid. so I have no better idea on how to solve this...

zerox1212, 2016-07-28:
I understand more or less now. I suppose the only option would be to add ```node.get_data_type_as_str()``` method, but I realize node already has many methods...

It isn't that important, when I was originally using it, it just happened to return something I didn't expect because I didn't fully understand the OPC UA spec.

zerox1212, 2016-08-23:
For reference the object id dict is used, but you need to import the module explicitly. The XML exporter can be used as an example, it uses the reverse look up.

Closing.

closed 2016-08-23 by zerox1212