FreeOpcUa/python-opcua #152: client PublishRequest timeout

oroulet, 2016-04-01:
check what timeout we send publishrequest with (9e8). Ideally we should use 0, try 0 again. Otherwise we need to use thread to resend publish requests that have timeouted on server.

This issue is solved on server side.

georgschoelly, 2016-08-12:
The spec Part 4, 7.28 states:

> The Server shall check the `timeoutHint` parameter of a `PublishRequest` before processing a `PublishResponse`. If the request timed out, a `Bad_Timeout` Service result is sent and another `PublishRequest` is used.

Doesn't that just mean that we can catch `BadTimeout` in the publish callback handler and re-send another publish request?

oroulet, 2016-08-12:
@georgschoelly yes, might be a solution. Although using 0 seems even better... if servers support it

georgschoelly, 2016-08-12:
I created a quick pull request, but I don't have a setup ready to test this. Do you have something where you could easily test it? If not, I'm going to give it a try later.

destogl, 2016-09-28:
After #282 this should be solved?!

georgschoelly, 2016-09-28:
@destogl: Not really. The code still uses `9e8` as the timeout. As oroulet says, `0` (= no timeout) would be better.  I don't know what the problem was with `0`, so I don't know if we can just replace `9e8` with `0`.

oroulet, 2016-10-01:
I think some servers had issues with 0. But we can try again, so we follow correctly spec. Send pr

ramirahikkala, 2017-11-14:
I'm working with Festo PLC and it has an OPC UA server that doesn't allow 9e8 as a timeout but works with 0.

closed 2017-11-17 by oroulet