FreeOpcUa/python-opcua #154: Malformed packages: CreateSessionResponse and ActivateSessionResponse

destogl, 2016-04-07:
By debugging of some errors between python-opcua and freeopcua libs I found out that recently we introduced a new bug... CreateSessionResponse and ActivateSessionRequest are not recognised by Wireshark but marked as malformed (s. picture below)

Can someone confirm this?

[image]

oroulet, 2016-04-07:
Probably the change to send null string instead of string of length 0. I
had to revert it once already. Do you see where the parsing fails? In the
box under

>

destogl, 2016-04-07:
`CreateSessionResponse` seams to break on/after `RevisedSessionTimeout`:

[image]

`ActivateSessionResponse` on/after `AdditionalHeader`:

[image]

oroulet, 2016-04-07:
I just checked on my pc and I do not have the issue. Using wireshark 2.0.1 and freeopcua master

destogl, 2016-04-08:
OK, I was using 1.10.6 version from Ubuntu repositories.

Install the version 2.0.2 from [stable ppa](https://launchpad.net/~wireshark-dev/+archive/ubuntu/stable) the error doesn't exist any more :+1: 

Sorry for false alarm, but it is good we cleared it!

closed 2016-04-08 by destogl