FreeOpcUa/python-opcua #157: client timeout error

oroulet, 2016-04-11:
I am using a python client to access a server written either with the open62541 stack or with FreeOpcUa python stack.
Both client and server are running on the same Raspberry Pi.
The server in both cases follows the minimal example hosting one double variable which I'm accessing.
The client visualizes this variable on a small OLED display.

For both servers, the client aborts after some run time, typically after a few hours.
The server is unaffected by this abort, parallel access with different clients is always possible.

The error messages I get when using the open62541 server are a little more informative :
First question - what is the token id, what is it used for and why could it be wrong?
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 93, in _run
    self._receive()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 100, in _receive
    msg = self._connection.receive_from_socket(self._socket)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/ua/uaprotocol_hand.py", line 648, in receive_from_socket
    return self.receive_from_header_and_body(header, utils.Buffer(body))
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/ua/uaprotocol_hand.py", line 620, in receive_from_header_and_body
    return self._receive(chunk)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/ua/uaprotocol_hand.py", line 651, in _receive
    self._check_incoming_chunk(msg)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/ua/uaprotocol_hand.py", line 582, in _check_incoming_chunk
    self._security_token.TokenId))
UaError: Wrong token id 2, expected 3

Traceback (most recent call last):
  File "display_client.py", line 33, in <module>
    val = setpoint.get_value()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 84, in get_value
    result = self.get_data_value()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 93, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/client.py", line 51, in run
    val = server_state.get_value()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 84, in get_value
    result = self.get_data_value()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 93, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 189, in get_attribute
    result = self.server.read(params)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 271, in read
    data = self._uasocket.send_request(request)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 75, in send_request
    data = future.result(self.timeout)
  File "/usr/local/lib/python2.7/dist-packages/futures-3.0.5-py2.7.egg/concurrent/futures/_base.py", line 407, in result
    raise TimeoutError()
TimeoutError
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 189, in get_attribute

    result = self.server.read(params)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 271, in read
    data = self._uasocket.send_request(request)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 75, in send_request
    data = future.result(self.timeout)
  File "/usr/local/lib/python2.7/dist-packages/futures-3.0.5-py2.7.egg/concurrent/futures/_base.py", line 407, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError

When running the client with the python server, the abort looks quite similar, the first line is the server message:
('Cleanup client connection: ', ('149.220.56.121', 56950))
Traceback (most recent call last):
  File "display_client_pyserver.py", line 33, in <module>
    val = setpoint.get_value()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 84, in get_value
    result = self.get_data_value()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 93, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 189, in get_attribute
    result = self.server.read(params)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 271, in read
    data = self._uasocket.send_request(request)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 75, in send_request
    data = future.result(self.timeout)
  File "/usr/local/lib/python2.7/dist-packages/futures-3.0.5-py2.7.egg/concurrent/futures/_base.py", line 407, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/client.py", line 51, in run
    val = server_state.get_value()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 84, in get_value
    result = self.get_data_value()
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 93, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/common/node.py", line 189, in get_attribute
    result = self.server.read(params)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 271, in read
    data = self._uasocket.send_request(request)
  File "/usr/local/lib/python2.7/dist-packages/freeopcua-0.10.9-py2.7.egg/opcua/client/ua_client.py", line 75, in send_request
    data = future.result(self.timeout)
  File "/usr/local/lib/python2.7/dist-packages/futures-3.0.5-py2.7.egg/concurrent/futures/_base.py", line 407, in result
    raise TimeoutError()
TimeoutError

It may be of interest to know the client code, it is not much in there but the minimal client example:
import oled_ssd1331 as oled
import datetime
import time
from opcua import Client

if __name__ == '__main__':

    # setup the display - define the pins
    device = oled.SSD1331(oled.SSD1331_PIN_DC, oled.SSD1331_PIN_RST, oled.SSD1331_PIN_CS)

    # setup the OPC-UA client - define host and port
    client = Client("opc.tcp://149.220.56.121:16664")

    try:
        device.EnableDisplay(True)
        device.Clear()

        # connect to the OPC-UA server
        client.connect()
        # browse to the phase setpoint variable node
        root = client.get_root_node()
        phaseshifter = root.get_child(["0:Objects", "1:PhaseShifter"])
        setpoint = phaseshifter.get_child("1:Setpoint")

        while True:

            # read and display the system time
            my_now = datetime.datetime.now()
            today_time = my_now.strftime("%H:%M:%S")
            device.DrawString(10, 5, today_time, oled.COLOR_WHITE, oled.COLOR_BLUE)

            # read and display the phase setpoint
            val = setpoint.get_value()
            valstr = '%9.4f' % val
            device.DrawString(10,25, valstr, oled.COLOR_WHITE, oled.COLOR_RED)

            time.sleep(0.25)

    except KeyboardInterrupt:
        print 'Interrupt caught. Exiting...'
        # close diplay
        device.EnableDisplay(False)
        device.Remove()
        # close OPC-UA connection
        client.disconnect()

alkor, 2016-04-11:
Looks like a bug in renewing a secure channel. I'll try to fix this issue.

oroulet, 2016-04-11:
Not sure. this is an error against openXXX stack. They are not exactly free
from bugs either :-)
But I have a collegue who says that he get disconnections when leaning a
client running for several days, so I think we have an issue somewhere...
in addtion to the know timeout issue of publishrequests (we should resend
publishrequests if they timeout)

oroulet, 2016-04-15:
@alkor https://github.com/FreeOpcUa/python-opcua/commit/f22a3f762e6e661fcefca019ce7c9b4d1e076e10 changes the way a null string is packed and unpacked. It seems to make some trouble with encryption. Could you have a look with master? I just fixed an error but I still cannot connect to a prosys server with encryption:
alHeader:None), DeleteSubscriptions:True)
Traceback (most recent call last):
  File "client_to_prosys.py", line 30, in <module>
    client.connect()
  File "../opcua/client/client.py", line 208, in connect
    self.open_secure_channel()
  File "../opcua/client/client.py", line 250, in open_secure_channel
    result = self.uaclient.open_secure_channel(params)
  File "../opcua/client/ua_client.py", line 219, in open_secure_channel
    return self._uasocket.open_secure_channel(params)
  File "../opcua/client/ua_client.py", line 160, in open_secure_channel
    response = ua.OpenSecureChannelResponse.from_binary(future.result(self.timeout))
  File "/usr/lib/python3.5/concurrent/futures/_base.py", line 407, in result
    raise TimeoutError()
concurrent.futures._base.TimeoutError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "client_to_prosys.py", line 67, in <module>
    client.disconnect()
  File "../opcua/client/client.py", line 217, in disconnect
    self.close_session()
  File "../opcua/client/client.py", line 421, in close_session
    return self.uaclient.close_session(True)
  File "../opcua/client/ua_client.py", line 253, in close_session
    data = self._uasocket.send_request(request)
  File "../opcua/client/ua_client.py", line 73, in send_request
    future = self._send_request(request, callback, timeout, message_type)
  File "../opcua/client/ua_client.py", line 63, in _send_request
    msg = self._connection.message_to_binary(binreq, message_type, self._request_id)
  File "../opcua/ua/uaprotocol_hand.py", line 534, in message_to_binary
    token_id=self._security_token.TokenId)
  File "../opcua/ua/uaprotocol_hand.py", line 435, in message_to_chunks
    max_size = MessageChunk.max_body_size(crypto, max_chunk_size)
  File "../opcua/ua/uaprotocol_hand.py", line 411, in max_body_size
    max_plain_size = (max_encrypted_size // crypto.encrypted_block_size()) * crypto.plain_block_size()
  File "../opcua/crypto/security_policies.py", line 123, in encrypted_block_size
    return self.Encryptor.encrypted_block_size()
AttributeError: 'NoneType' object has no attribute 'encrypted_block_size'

jfapr85, 2016-05-26:
I have exactly this problem using subscriptions to datachange_notification. I'm using it against freeopcua C++ Server using unsafe mode. 

I check the version that I've installed and it is previous to #f22a3f7 (the last available in pypi). Is there a solution for this problem? Given that I'm using unsafe connection, would be the downgrade a solution for this particular problem?

Thanks!

oroulet, 2016-05-26:
This is an interesting information. The freeopcua c++ server does not check duration of session. So this is either a network error or an error within the client somewhere....

jfapr85, 2016-05-26:
In the log there are traces similar to that one when it stopped working:

Exception in thread Thread-3:
Traceback (most recent call last):
File "/home/user/anaconda2/lib/python2.7/threading.py", line 801, in __b$
self.run()
File "/home/user/anaconda2/lib/python2.7/threading.py", line 754, in run
self.__target(*self.__args, **self.__kwargs)
File "/home/user/anaconda2/envs/adapt/lib/python2.7/site-packages/opcua/$
self._receive()
File "/home/user/anaconda2/envs/adapt/lib/python2.7/site-packages/opcua/$
msg = self._connection.receive_from_socket(self._socket)
File "/home/user/anaconda2/envs/adapt/lib/python2.7/site-packages/opcua/$
return self.receive_from_header_and_body(header, utils.Buffer(body))
File "/home/user/anaconda2/envs/adapt/lib/python2.7/site-packages/opcua/$
return self._receive(chunk)
File "/home/user/anaconda2/envs/adapt/lib/python2.7/site-packages/opcua/$
self._check_incoming_chunk(msg)
File "/home/user/anaconda2/envs/adapt/lib/python2.7/site-packages/opcua/$
self._security_token.TokenId))
UaError: Wrong token id 2, expected 3

jfapr85, 2016-05-31:
I have tried against a python freeopcua server and it remains happening, so I suppose it is a bug on the client side. ¿Any idea of how can solve it or at least catch the exception in order to try a reconnection?

Thanks!

oroulet, 2016-05-31:
Hi Anyway if you want to ensure stable connection you need to poll the server, since we do not get any waning in case of network error until we try to renew session or secure channel. This is what all clients do.
something like:
```

   c = client(myurl)
   while True:
        c.connect()
       test_node = c.get_node(xxxxx)
        while true:
             time.sleep(x)
              try:
                    test_node.get_value()
               except:
                     break
```

But we have a bug somewhere, and I would like to find out where.... no idea how

oroulet, 2016-06-12:
thinking about it looks like we received a packet with an ole TokenId after the server increased it. It seems reasonable that it sometimes happen due to network and mutithreading. I created PR https://github.com/FreeOpcUa/python-opcua/pull/206 that will simply ignore such messages. now the exception will happen in main client thread, but this should be handled by user anyway since exceptions might happen over network. Accepting wrong TokenId is the other alternative, but it sounds like a very bad idea...

oroulet, 2016-07-26:
probably solved

closed 2016-07-26 by oroulet