FreeOpcUa/python-opcua #159: Autogenerate the UATypes

destogl, 2016-04-19:
Following the discussion in #133 I saw that we have problem with not to generate UA Types (standard types) as python objects. This can be very useful to achieve good mapping between UA Types and internal objects. So my proposal would be following:

* For every UA Type we should have mapping to Python types/objects (this is already achieved in `uatypes.py`
* Every UA Type could extend Python type/object to have all properties as python type/object, e.g. on `BaseEvent` UA Object

   `BaseEvent` (ua) should inherit `Event` (python) which than has `fill_base_fileds`-mehtod.

In this way `BaseEvent` class can be autogenerated  and the same is true for all other (inheriting) events...

* Similar approach is also interesting for other types/object like base types or Node/NodeId...

This is just my two cents regarding this issue.... hopefully at one point in time I provide also some inputs regarding that.

oroulet, 2016-04-19:
OK let's try, if you make it we can merge your event stuff as is :-)
But generate the objects in their own files! and maybe only start with events. And I also propose to create a new python file to parse xml and generate types so we do not make the generate_address_space.py too complicated. We need to do something completely different...

oroulet, 2016-04-19:
OK I wrote some code for it. I will show it soon BUT I am sure this is not enough for events.
Typically a server does not generate BaseEvents it generate custom events from its custom datamodell. You want to let user to create new events in python something like

```
basevent = server,get_node(ua.ObjectIds.BaseEventType)
myeventtype = basevent.add_object("MyNewEvent")
myeventtype.add_property("MySuperProperty", 9,8)
myeventtype.add_property("MySuperOtherProperty", 9,8)

# now use my event
generator = server.get_event_generator(myeventtype)
generator.MySuperProperty= 4
generator.trigger()
```

And on client side
```
customeventype = client.get_node(NodeIDOfEvent)
sub = client.create_subscription()
sub.subscribe_events(customeventtype)
```

destogl, 2016-04-19:
I will just write one small proposal...

For custom events I agree with you, lets do this. Even though I am more fun of writing new event class with all properties.... But actually we can enable booth of this things... Just give me a second :)

closed 2016-06-12 by oroulet