FreeOpcUa/python-opcua #160: BadDataEncodingInvalid

dualmal, 2016-04-19:
I'm working on trying to get reading tag values out of my OPC server using a python client. I can connect, authenticate via user/pass, browse tags, get their children and descriptions. That all works great.

Problem: Whenever I invoke a function which uses the "get_attribute" method on a Node object I get a "BadDataEncodingInvalid" error. Full stacktrace is attached below. My server does work. It is Inductive Automation's OPC UA server. I can verify its working because I've been able to read (and write) data from it with multiple GUI clients, as well as the Ignition Gateway which is running my whole SCADA layer. 

So I'm wondering what I'm doing wrong (or need to do) in the client. A sample of what I'm trying to do it also below.

Thanks everyone!
`
>>> walltime = client.get_node('ns=1;s=[Line_PLC1]Global.WallTime')

>>> walltime
Node(StringNodeId(ns=1;s=[Line_PLC1]Global.WallTime))

>>> walltime.get_children()
[Node(StringNodeId(ns=1;s=[Line_PLC1]Global.WallTime.Day)), Node(StringNodeId(ns
=1;s=[Line_PLC1]Global.WallTime.Second)), Node(StringNodeId(ns=1;s=[Line_PLC1]Gl
obal.WallTime.Year)), Node(StringNodeId(ns=1;s=[Line_PLC1]Global.WallTime.Micros
econd)), Node(StringNodeId(ns=1;s=[Line_PLC1]Global.WallTime.Month)), Node(Strin
gNodeId(ns=1;s=[Line_PLC1]Global.WallTime.Minute)), Node(StringNodeId(ns=1;s=[Li
ne_PLC1]Global.WallTime.Hour))]

>>> a = client.get_node('ns=1;s=[Line_PLC1]Global.WallTime.Second')

>>> a
Node(StringNodeId(ns=1;s=[Line_PLC1]Global.WallTime.Second))

>>> a.get_data_value()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "opcua\common\node.py", line 98, in get_data_value
    return self.get_attribute(ua.AttributeIds.Value)
  File "opcua\common\node.py", line 205, in get_attribute
    result[0].StatusCode.check()
  File "opcua\ua\uatypes.py", line 375, in check
    raise UaStatusCodeError("{}({})".format(self.doc, self.name))
opcua.common.uaerrors.UaStatusCodeError: The data encoding is invalid.(BadDataEn
codingInvalid)
`

oroulet, 2016-04-19:
> File "", line 1, in

Can you register a session with wireshark? So that we understand what is
happening

dualmal, 2016-04-20:
Attached is the PCAP of the console session I just ran. Console code also attached. 

[baddataencodinginvalid.zip](https://github.com/FreeOpcUa/python-opcua/files/228381/baddataencodinginvalid.zip)

zerox1212, 2016-04-20:
Have you tried subscribing to ```a``` instead of calling ```get_data_value()``` directly?

oroulet, 2016-04-21:
try this:
https://github.com/FreeOpcUa/python-opcua/pull/162

dualmal, 2016-06-01:
Sorry I never reported. This did solve the issue. Thanks for your prompt response. Its exciting to see this being consistently maintained.

closed 2016-06-01 by dualmal