FreeOpcUa/python-opcua #169: Logging

Adirio, 2016-04-26:
Subscriptions are creating a new logger without any handler. Is this intended? I was getting `Received a notification for unknown handle: XXX` message and wasn't seeing it. Instead I was seeing: `No handlers could be found for logger "opcua.common.subscription"`

By the way, this message may be logged just after unsubscribing from a variable depending on the period.

oroulet, 2016-04-26:
This message is fine. Some servers send reply to all publishrequest in queue AFTER the subscription has ended. We can safely ignore then. The missing handler stuff is a bit strange....

destogl, 2016-04-26:
Yes I get also often this handler error, but never looked closer at it... maybe it is time to solve it :) I will give also feedback, when I see it again.

oroulet, 2016-04-27:
Strange I have never had it..... And I run many clients / servers in many threads

oroulet, 2016-04-27:
Can it be that you do not configure logging in you own code? If so then it is harmless, but should just logging :-)

destogl, 2016-04-27:
This is possible :)
Can we do some defaults when user is not configuring it?

oroulet, 2016-04-27:
No we cannot. This is as designed by logging module people. Trying to bypass this leads to many issues.... I still have a colleague doing this and this is a nightmare

destogl, 2016-04-27:
OK. So

@Adirio:  I guess we can close the issue?

closed 2016-04-28 by oroulet