FreeOpcUa/python-opcua #173: pip install freeopcua – !!! WARNING !!! The Trollius project is now deprecated!

koppi, 2016-05-01:
```bash
# pip install freeopcua
Downloading/unpacking freeopcua
  Downloading freeopcua-0.10.11.tar.gz (256kB): 256kB downloaded
  Running setup.py (path:/tmp/pip-build-iNghtC/freeopcua/setup.py) egg_info for package freeopcua
    
Downloading/unpacking enum34 (from freeopcua)
  Downloading enum34-1.1.4-py2.py3-none-any.whl (61kB): 61kB downloaded
Downloading/unpacking trollius (from freeopcua)
  Downloading trollius-2.1.tar.gz (276kB): 276kB downloaded
  Running setup.py (path:/tmp/pip-build-iNghtC/trollius/setup.py) egg_info for package trollius
    !!! WARNING !!! The Trollius project is now deprecated!
    
    
    warning: no files found matching 'release.py'
    
    !!! WARNING !!! The Trollius project is now deprecated!
Downloading/unpacking futures (from freeopcua)
  Downloading futures-3.0.5-py2-none-any.whl
Requirement already satisfied (use --upgrade to upgrade): six in /usr/lib/python2.7/dist-packages (from trollius->freeopcua)
Installing collected packages: freeopcua, enum34, trollius, futures
  Running setup.py install for freeopcua
```
Can I ignore this warning?

zerox1212, 2016-05-02:
You can ignore it, but I strongly recommend developing in Python 3.4.

koppi, 2016-05-02:
oki, closing then..

closed 2016-05-02 by koppi