FreeOpcUa/python-opcua #174: Issue with Subscribing to Multiple Event Types

zerox1212, 2016-05-02:
As I have been writing history support for events I found this issue.

If I set up two custom event types with event generators on the same object node, both event objects return the same properties.

See screen shots:
[image]

Event results that show up in SubHandler.event_notification
Object 2001, EventType 2003
[image]

Object 2001, EventType 2006
[image]

Code (event_types comes from your example in https://github.com/FreeOpcUa/python-opcua/pull/171 )
```python
ev_handlers = []
for event_type in event_types:
    handler = self._sub.subscribe_events(source, event_type)
    ev_handlers.append(handler)
self._handlers[source] = ev_handlers
```

destogl, 2016-05-02:
This is the problem you mentioned in #172. This depends not on event implementation but on subscription implementation. so we should take deeper look into spec and maybe reimplement subscriptions...

oroulet, 2016-05-02:
"reimplement" is a big word, but we surely need to fix this. I am guessing
this is due to the WhereClause which we currently ignore.
Btw does subscribing to events with uaexpert work currently? I tried
yesterday with your server-event.py file and I could not see anything in
uaexpert.,,,,

zerox1212, 2016-05-02:
If you take the code from my PR it works in UaExpert. However I think you need to set the start and end times a day early and a day late. I'm not sure what time zone it uses.

destogl, 2016-05-02:
For me worked without any issues.

oroulet, 2016-05-02:
ok I tested again. it works, I was just confused because in the example file you make the Object node generate events. This is unusual, I have never seen a server emit events from another node than Server. Did you? If not maybe we should also generate events from sever node in example.

another thing is that the common/event.py file now only contains server code. maybe you should move it?

also sourceName property seems to be a binary string in uaexpert. Is this normal?

And our answer to a modifyMonitoredItemRequest seems to be broken. Wireshark does not manage to parse it

destogl, 2016-05-03:
>This is unusual, I have never seen a server emit events from another node than Server. Did you? If not maybe we should also generate events from sever node in example.
Yes I saw servers generating events from its objects. E.g. what would you do if you have multiple instances of on ObjectType representing different machines or multiple objects of different types representing parts of a machine, and every one of them can generate events (possibly the same type). If you use Server as source than you lost connection between the event and the object and than you can properly (and fast) react to the events.

> another thing is that the common/event.py file now only contains server code. maybe you should move it?
Yes, common/event.py contains **mostly** server code. Only exception is the `get_event_properties_from_type_node` which is used by common/subscription.py to crate an event filter.

>also sourceName property seems to be a binary string in uaexpert. Is this normal?
According to specs (Part 5, section 6.4.2, table 22) `SourceName` is of type `String` so I guess that binary string is OK.

>And our answer to a modifyMonitoredItemRequest seems to be broken. Wireshark does not manage to parse it
OK, this should be looked at. I didn't do anything regarding that. I hope you are using the recent version of Wireshark since I  had problems with older version (the object were not correctly described).

Where are we using `modifyMonitoredItemRequest`?

oroulet, 2016-05-03:
> Yes, common/event.py contains mostly server code. 

Can you move server code on server siden then? You have now better overview than I have.

> Yes I saw servers generating events from its objects.

OK then we keep it as is. As you say it make sense.

> And our answer to a modifyMonitoredItemRequest seems to be broken.

https://github.com/FreeOpcUa/python-opcua/issues/175

Next thing is that we need to understand how to filter by events. As I wrote somewhere else and I cannot make sense of the EventFilter stuff. One place or another, there should be a way to filter by event type. But the UA spec is difficient many places so maybe this is another place

destogl, 2016-05-03:
> Can you move server code on server siden then? You have now better overview than I have.

I did it, but gitlab decided to block my organisation, hopefully the issue will be resolved soon and than I create PR.

I will check the specs regarding filtering... but it will take few days...

zerox1212, 2016-05-04:
I read the spec and now I understand what you mean. The subscription should actually be supplied with an EventFilter, not an event/event type. The EventFilter is used to allow event notifications to the client based on the SelectClauses and WhereClauses. If the event isn't in the filter the client doesn't get notified.

I see where you abstract an EventFilter from event type in the creation of the subscription. Then all the properties of that event are added to the filter. From my short look at the code, the problem described above looks more like a bug than an incorrect implementation (the filter should have the event type + all the properties of the event, it's just that the wrong result is returned).

For the work I'm doing with history this isn't too important. I do not know if we even need this server history to filter events. I would say that enabling history on a node in the server should always historize all events, so the only problem I have is the missing custom fields.

oroulet, 2016-05-04:
>> I read the spec and now I understand what you mean. 

great

>>  The subscription should actually be supplied with an EventFilter according to the spec, not an event/event type.

Well, do not take the spec as a perfect thing, it is over engineered and of full archaic things probably ported from legacy systems. I believe that putting events as parameter is a good abstraction. If a user really want to use the EventFilter stuff they can create a monitoreditem by han using the Subscription class.

zerox1212, 2016-05-04:
Agreed. My guess is that when I trigger `create_subscription` two times, only one EventFilter is created. If I took a guess, I would say `create_subscription` needs to take a LIST of event types, then build a single EventFilter for the node.

oroulet, 2016-05-04:
>> I would say create_subscription needs to take a LIST of event types

Yes we need to change something here. But before writing code I would like to get a good understanding of the EventFilter stuff, I still cannot understand the WhereClause well

zerox1212, 2016-05-04:
>Yes we need to change something here. But before writing code I would like to get a good understanding of the EventFilter stuff, I still cannot understand the WhereClause well

This is fine for me. I would like to add WhereClause support to SQL as well. Would be good to undertand it well. For now I just think about it as if it were a "WHERE" SQL statement, but I haven't checked the details yet.

oroulet, 2016-05-22:
a fix was merged in master, but it is not very well tested...

zerox1212, 2016-05-23:
I'll grab the master and update event history. Thanks.

zerox1212, 2016-05-24:
Check https://github.com/FreeOpcUa/python-opcua/pull/189 .

closed 2016-06-12 by oroulet